MichaIng/DietPi #6688: Flash the SPI bootloader on ROCK 4

andreagdipaolo, 2023-10-17:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

- I saw you introduced the ability to flash the SPI bootloader on the ROCK 5B on the upcoming 8.23 release

#### Describe the solution you'd like:

- It would be nice for this feature to be available for the ROCK 4 too

#### Additional context

- Flashing the SPI bootloader on the ROCK 4 is already a quite easy procedure thanks to the Armbian distro, which has a simple flasher (and their own flavour of U-Boot, frequently updated and different from the one baked by Radxa and the official procedure)
- You can find their solution [here](https://forum.armbian.com/topic/16195-booting-rock-pi-4abc-with-mainline-u-boot-in-spi-nvme-and-armbian-v2011x/)

Thanks!

MichaIng, 2023-10-18:
Just to be sure, this is the MTD device, right?
```sh
ls -l /dev/mtdblock0
```
And this file exists:
```sh
source /usr/lib/u-boot/platform_install.sh
ls -l "$DIR/rkspi_loader.img"
```
Hence this command(s) to flash the SPI bootloader to the MTD device:
```sh
source /usr/lib/u-boot/platform_install.sh
dd if="$DIR/rkspi_loader.img" of=/dev/mtdblock0 oflag=direct conv=notrunc status=progress
```

andreagdipaolo, 2023-10-18:
> Just to be sure, this is the MTD device, right?

I can't test it right now, but last time I performed that guide it was reliable and it was **three months ago**.
In order to do that I ran Armbian via usb or SD (I don't remember) so I believe I can't do that with DietPi without downloading some software and file.

I cannot see it with that command, so the overlay could be necessary.

> And this file exists

It surely did on Armbian if it worked, I believe.

> `source /usr/lib/u-boot/platform_install.sh`

I found this

> `ls -l "$DIR/rkspi_loader.img"`

I'm unsure about this one

> The thread you linked is ancient

Yep, but it still works as far as this summer, at least on their os

> What I am wondering is whether the device tree overlay (first step) is still required.

I believe so, because without the overlay at the moment I can't see that partition

> Not all have it, is it?

Exactly. Some have the pads only and can be populated. Some have it preinstalled (like mine). Some - the newest - don't have it and it can't be soldered either.

MichaIng, 2023-10-18:
Added with: https://github.com/MichaIng/DietPi/commit/18d660a

It is done with a little care, checking for both, the MTD device and the expected SPI U-Boot image. If no MTD device is found, it offers to apply the overlay. If after a reboot still no MTD device has been found, the ROCK 4 model seems to have none.

Would be great if you could test it via our `dev` branch:
```sh
G_DEV_BRANCH dev
dietpi-config 3
```

andreagdipaolo, 2023-10-18:
Will do that tonight. Thanks!
Have you found anything in their repo about bootloader changelogs?
At the moment I believe I have the one from June. It's not giving me any problem with my SSD.

andreagdipaolo, 2023-10-18:
> Would be great if you could test it via our dev branch

[image]

Reboot prompted while exiting the menu, done, but the device has not appeared and dietpi-config is unable to update SPI.
Maybe you missed this step of the guide:
`param_spinor_spi_bus=1` in the env file

Because I only see
`[  OK  ] DietPi-Config | Setting in /boot/dietpiEnv.txt adjusted: overlays=rk3399-opp-2ghz pcie-gen2 spi-jedec-nor`

I added it manually and now it appears, but after running `echo spi1.0 > /sys/bus/spi/drivers/spi-nor/bind` I receive `bash: echo: write error: Device or resource busy`.

It happened to me sometimes but I don't remember how I fixed that.

However, via dietpi-config it worked flawlessly 😃 

```
[ INFO ] DietPi-Config | dd if=/usr/lib/linux-u-boot-current-rockpi-4b/rkspi_loader.img of=/dev/mtdblock0 oflag=direct conv=notrunc status=progress, please wait...
3998208 bytes (4.0 MB, 3.8 MiB) copied, 26 s, 153 kB/s
8128+0 records in
8128+0 records out
4161536 bytes (4.2 MB, 4.0 MiB) copied, 27.2783 s, 153 kB/s
[  OK  ] DietPi-Config | dd if=/usr/lib/linux-u-boot-current-rockpi-4b/rkspi_loader.img of=/dev/mtdblock0 oflag=direct conv=notrunc status=progress
```

You just have to add the `param_spinor_spi_bus=1` line in the dietpiEnv.txt

How will I switch to release versions when 8.23 will be stable? I can wait, I don't like to downgrade.

MichaIng, 2023-10-18:
> You just have to add the `param_spinor_spi_bus=1` line in the dietpiEnv.txt

Interesting that a U-Boot environment variable is required, rather than a kernel command-line parameter (`bootargs=`) to be used by the kernel. Will add it as config step. It will still check for the overlay only, as the logic otherwise becomes too complicated for my brain tonight. But as with a stable release, both settings are then applied one after another, it should be fine.

andreagdipaolo, 2023-10-18:
I just put it there, at the end.
                                                                                          
```
rootdev=UUID=mindyourbusinesslol
rootfstype=ext4
# The init system logs to the console defined last.
consoleargs=console=tty1
usbstoragequirks=
extraargs=net.ifnames=0
docker_optimizations=on
overlay_path=rockchip
overlay_prefix=rockchip
overlays=rk3399-opp-2ghz pcie-gen2 spi-jedec-nor
user_overlays=
param_spinor_spi_bus=1
```

About retutning to the release channel, is it simple and i will not lose anything, right? As soon as I do it when 8.23 is out.

MichaIng, 2023-10-18:
DietPi-Config will now do:
```sh
G_CONFIG_INJECT 'param_spinor_spi_bus=' 'param_spinor_spi_bus=1' /boot/dietpiEnv.txt
```
I also pushed this to the `beta` branch now:
- #6691
- #6687

closed 2023-10-21 by MichaIng