MichaIng/DietPi #6689: Install DiePi In Orange Pi One

actraiser30, 2023-10-18:
Hello im trying to install DietPi in my Orange Pi One Board Using the latest Armbian Minimal Bookwom Image

and i get an error upon installation,that could not install some packages i press retry but is doing nothing like there is no internet but there is

before in Armbian bullseye image i was doing this when getting this error and  it was working after

i opened a  subshell and type inside:

echo 'iface eth0 inet dhcp' > /etc/network/interfaces
ifup eth0
exit

now when i type ifup eth0 it says command not found

any help??

thanks

MichaIng, 2023-10-18:
Can you show the full error message, at best the full output of the script? There should be nothing stopping the network connection with the minimal image, but probably they recently changed something.

> now when i type ifup eth0 it says: eth0 not found

Because Armbian uses "predictable" interface naming, hence it is not called `eth0` but something like `enp0s1`. You can check the names via:
```sh
ip l
```

actraiser30, 2023-10-18:
now i deleted the card,i tried multiple times and tired,in armbian bullseye the eth0 was working,i will try again and let you know

MichaIng, 2023-10-18:
We create images from scratch via `debootstrap`, so no pre-image is used. See our build script: https://github.com/MichaIng/DietPi/blob/master/.build/images/dietpi-build

actraiser30, 2023-10-18:
How can I create an image my self are there instructions anywhere??
I mean you need to boot an image to create this image right??

MichaIng, 2023-10-18:
> I mean you need to boot an image to create this image right??

You can create this on any Debian or Ubuntu host system. BUT: It does only work for supported SBCs, hence not the Orange Pi One. So in your case you need to keep using the `dietpi-installer`. There was actually an issue regarding the `base-files` package.  This has been fixed in the `dev` version of the installer:
```sh
bash -c "$(curl -sSfL 'https://raw.githubusercontent.com/MichaIng/DietPi/master/.build/images/dietpi-installer')"
```
If it still fails, it would be helpful it you retried it with the Armbian Bookworm image, and pasted the whole output of the script here + when you run into the network connection issue, from the subshell:
```sh
ip a
ip r
```

actraiser30, 2023-10-18:
Can't you create a one time only Image for the Orange Pi One ??
I know it's not in your plans to have an image for this board,but I think a
one time image can be possible without plans for an updated version just
this once,I also have a Cubieboard 2,I can't find in the installer list the
generic Allwinner A20 is it possible to add it???

Thanks

MichaIng, 2023-10-18:
> Can't you create a one time only Image for the Orange Pi One ??

Not without implementing support for it into all related scripts. There are a lot of other newer boards, including new ones from Orange Pi, which we'll implement with higher priority. There is no point for this effort without aiming to keep it maintained as well.

For the Cubieboard, try to use the "Generic Device" option.

actraiser30, 2023-10-19:
The command ip 1 returns nothing
And the eth0 it's not the enp0s1
Any idea how can I found out what name has the Ethernet adapter??

Joulinar, 2023-10-19:
can you show output of

```
ip a
ip r
```

actraiser30, 2023-10-19:
[image]

actraiser30, 2023-10-19:
When I do ifup end0 it says device does not exists

MichaIng, 2025-05-31:
Sorry for the missed reply. In case this is still a question: as the interface is called `end0` now with "predictable" (obviously changing between systemd versions) interface names, for **e**mbedded **n**etwork **d**evices. So in the subshell, this would now be required:
```sh
echo 'iface end0 inet dhcp' > /etc/network/interfaces
ifup end0
exit
```
Still didn't find time to implement handling of NetworkManager uninstalls, which tend to deconfigure the current network connection, for whatever reason. Though not always, on own tests, they remained active.