MichaIng/DietPi #6695: dietpi-update failing

theharshl, 2023-10-19:
### Creating a bug report/issue

#### Details:
- Date           | Thu Oct 19 14:01:58 MDT 2023
- DietPi version | v8.19.1 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS (32-bit) Lite
- Hardware       | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | `Linux trailer-server 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `systemctl start systemd-timesyncd`
- Exit code      | 5
- Software title | DietPi-TimeSync
#### Steps to reproduce:

1. execute dietpi-update
#### Expected behaviour:

- Update should run
#### Actual behaviour:

- Update does not run
#### Extra details:

- This may stem from my upgrade distros in place rather than reimaging. I wasn't excited about trying to recreate all my customizations and restoring application backups for things like Emby.
#### Additional logs:
```
Failed to start systemd-timesyncd.service: Unit systemd-timesyncd.service not found.
```

Joulinar, 2023-10-20:
I guess similar to this one https://github.com/MichaIng/DietPi/issues/6381#issuecomment-1551153725

theharshl, 2023-10-20:
Yep, sorry, not sure how I overlooked this in my search before submitting. I have run the commands referenced there and is fixed.

closed 2023-10-20 by theharshl