MichaIng/DietPi #6699: Lookup has failed.

TheGitGuy007, 2023-10-22:
Seems to be a invalid link here
[image]

Command: curl -ILfvm 10 https://keyserver.ubuntu.com/pks/lookup?search=0x2009837CBFFD68F45BC180471F4F90DE

Unsure what it's related too sadly.

Joulinar, 2023-10-22:
I had a look, and it seems `curl` is returning a HTTP 405 error. Something we need to have a look into

```
< HTTP/1.1 405 Method Not Allowed
HTTP/1.1 405 Method Not Allowed
```

However, you can also skip this step to work around the problem. If you have run the installation process and the error screen is displayed, you should have an option to change the command. Select this option and remove the entire command specified. Replace it with `true` and continue.

MichaIng, 2023-10-22:
The keyserver does not allow HEAD requests (the `-I` flag of `curl`), only GET (regular download). So we must remove the URL check. I haven't seen such case for a very long time.

Fixed with: https://github.com/MichaIng/DietPi/commit/c309978

I'll add a live patch as well.

MichaIng, 2023-10-22:
Live patch merged. Should be offered when running `dietpi-update`.

closed 2023-10-22 by MichaIng