MichaIng/DietPi #6700: dietpi-update error

RedNo7, 2023-10-22:
- Date           | Sun Oct 22 12:17:57 BST 2023
- Bug report     | 4976c6df-ed57-426d-a1d3-060120c3f205
- DietPi version | v8.22.3 (MichaIng/master)
- Image creator  | 
- Pre-image      | 
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux PetitPois 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bookworm (ID=7,RASPBIAN=0)
- Command        | `curl -sSfLO https://github.com/MichaIng/DietPi/archive/master.tar.gz`
- Exit code      | 35
- Software title | DietPi-Update
#### Steps to reproduce:

1. ran `diet pi-update`
2. see error log below:

#### Expected behaviour:
it should have worked?

#### Actual behaviour:
I got an error and was told to copy/paste the blue lines here.

#### Additional logs:
```
curl: (35) Send failure: Connection reset by peer

Joulinar, 2023-10-22:
It's connection to GitHub failing. Can you try again! Maybe a temporary issue.

RedNo7, 2023-10-22:
same error.  I'm off to lunch so can try again in a few hours.

Joulinar, 2023-10-22:
I had it as well once. But repeating the update a minute later succeeded. Maybe GitHub has some challenges on weekends?

RedNo7, 2023-10-22:
OK, worked just now 👍

closed 2023-10-22 by RedNo7