MichaIng/DietPi #6703: Installation error of Kodi: post-installation script fails

joshi0531, 2023-10-22:
### Creating a bug report/issue

#### Required Information
- DietPi version 
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=23
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
- Distro version "bookworm 0"
- Kernel version "Linux kkodi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux"
- SBC model "RPi 4 Model B (aarch64)"
- Power supply used  "5V 2A Nintendo switch power supply"
- SD card used "usb sata m.2 ssd"

#### Additional Information (if applicable)
- Software title  "Samba-client, FFmpeg, youtube-dl, Python 3 pip"
- clean install of the dietpi OS
- yes with every fresh installation of the dietpi os or directly after the auto setup.

- Bug report ID "7d60558b-02b6-4c6a-8aa2-ff83f621b8c2"

#### Steps to reproduce

1. Install Kodi with the command `sudo apt-get install kodi` or over the dietpi-software ui.
2. The error occurs when the post-installation script is executed.

#### Expected behaviour
Kodi should be successfully installed and configured.

#### Actual behaviour
The error "dpkg: error processing package kodi (--configure): installed kodi package post-installation script subprocess returned error exit status 1" occurs.

#### Extra details
- DietPi Version: [insert your DietPi version here]
- Kodi Version: 3:20.2-2~bookworm
- Excerpt from `dpkg --configure -a` log:

  Setting up kodi (3:20.2-2~bookworm) ...
  dpkg: error processing package kodi (--configure):
   installed kodi package post-installation script subprocess returned error exit status 1
  Errors were encountered while processing:
   kodi

- Excerpt from the post-installation script:

  sudo sh -x /var/lib/dpkg/info/kodi.postinst configure
  + set -e
  + [ configure = configure ]
  + which update-menus
  + [ -x ]
  + [ ! -f /etc/sudoers.d/020_kodi ]
  + [ ! -f /etc/udev/rules.d/99-kodi.rules ]
  + [ ! -f /etc/polkit-1/localauthority/50-local.d/kodi.pkla ]
  + tee -a /etc/polkit-1/localauthority/50-local.d/kodi.pkla

MichaIng, 2023-10-22:
Strange, this looks like this ancient error which happened only with the 32-bit package on Buster: https://github.com/RPi-Distro/repo/issues/153

Ah, indeed the missing directory is not even shipped anymore on Bookworm, as it has been obsolete. So there is an updated required now by the RPi maintainers.

MichaIng, 2023-10-24:
If `kodi-repository-kodi` was installed, then you had a very different issue and faced another error 😄. However, if on v8.22 Kodi was installed from the Debian repo along with `kodi-repository-kodi`, and with v8.23 the RPi repo is moved to Bookworm and then the `kodi` package from the RPi repo installed on top, this might indeed conflict, as both packages ship the same files. I guess you do not have the error message anymore, do you?

joshi0531, 2023-10-24:
it had been a cleaner installation because I wanted to set up kodi again and then came these errors that, the autoinstall has completely scraped. i have then made it with after install and have taken as help chatgpt. now everything walks, but that can still change, after which it looks but not yet.

MichaIng, 2023-10-24:
@joshi0531 
I was addressing leckye only 😉. Your (different) issue is a bug in the RPi Kodi package, which I linked and the workaround I posted above.

MichaIng, 2023-11-07:
No problem. This commit should solve the package conflict, so Kodi can upgrade to the build from the RPi repo without file conflicts: https://github.com/MichaIng/DietPi/commit/5365581

natevns08, 2023-12-13:
> Strange, this looks like this ancient error which happened only with the 32-bit package on Buster: [RPi-Distro/repo#153](https://github.com/RPi-Distro/repo/issues/153)
> 
> Ah, indeed the missing directory is not even shipped anymore on Bookworm, as it has been obsoleted. So there is an update required now by the RPi maintainers.
> 
> Workaround:
> 
> ```shell
> sudo mkdir -p /etc/polkit-1/localauthority/50-local.d
> ```

Wanted to throw my name in the hat, this worked for me as well.

MichaIng, 2023-12-14:
@freemanpt 
Since the hardware acceleration issue is unrelated to the installation issue, can you please open a new issue about this?

I re-added the workaround for Bookworm to `dietpi-software`.

closed 2023-12-18 by MichaIng