MichaIng/DietPi #6705: "usr-is-merged package cannot be installed" error installing DietPi on s905x box with Armbian - Bookworm

Black6spdZ, 2023-10-24:
get this error during install:
******************************************************************************
*
* The usr-is-merged package cannot be installed because this system does
* not have a merged /usr.
*
* Please install the usrmerge package to convert this system to merged-/usr.
*
* For more information please read https://wiki.debian.org/UsrMerge.
*
******************************************************************************

dpkg: error processing archive /tmp/apt/archives/usr-is-merged_35_all.deb (--unpack):
 new usr-is-merged package pre-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 /tmp/apt/archives/usr-is-merged_35_all.deb

Running this command solves the issue:
/usr/lib/usrmerge/convert-usrmerge

MichaIng, 2023-10-24:
Many thanks for reporting.

From where did you get an image that is not usr-merged? This is mandatory for Debian Bookworm, so all Bookworm images should be usr-merged, or considered as broken. All official Armbian images are definitely usr-merged.

Pooort, 2024-03-07:
Thank you for quick response. 

It was official image: `docker pull apache/airflow:2.7.2-python3.9`

I've just tried with the latest 2.8.2. It's completely ok.

Thank you a lot.

MichaIng, 2024-03-07:
Are you sure you are answering to the right issue? This Docker image does not use DietPi, but a Debian-based minimal Python image. And I can only hope that no one is trying to run the DietPi installer within a Docker container forged to serve a particular single application 😄.

Anyway, marking this as closed, as OP did not respond.

closed 2024-03-07 by MichaIng