MichaIng/DietPi #6708: Custom scene in Dietpi-Cloudshell (by looping through variable/array or files which can contain scenes)

roelbroersma, 2023-10-25:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:
An example: We use openvpn on DietPi and there is NO scene for openvpn (like connected/disconnected, clients, etc.) As well on the clients as on the server.

#### Describe the solution you'd like:
I would like to program a DietPi Cloudshell scene for OpenVPN, however, to make it more generic let's discuss a 'custom scene integration'.
I guess there are other people also wanting custom scenes.
In have thought about two possibilities:

1) Having a variable which can contain an array of custom_scene_files,  like:   DIETPI_CLOUDSHELL_CUSTOM_SCENES="/home/dietpi/openvpn_scene.sh,/home/dietpi/vu_meter_scene.sh"

2) Having a variable or specific folder which is enumerated for files (the files are the scenes).
E.g., so we can put this in the variable: /home/dietpi/cloudshell/scenes/* 

#### Describe alternatives you've considered:
- Modifying the Cloudshell for OpenVPN and do a pull/push/merge
- Creating our own solution based on a Python adafruits library and forget Cloudshell.

#### Additional context
I can help/do the programming. But I want to discuss this first because I don't want to do work that nobody wants ;)

MichaIng, 2023-10-26:
Many thanks for your suggestion.

Generally I like the idea of custom scenes, especially since this implies some generic way of defining and loading scenes. This however means also some infrastructural rework of the script, to combine all functions and variables which belong to a script in a way which allows to add more, likely an array. Ah, basically what you say already. Whether the array(s) are then extended with a single file, or multiple files within a directory, should not matter much. The latter would allow easier extension. But please do not use the `dietpi` user's home dir, as this user will go eventually. Instead I suggest `/var/lib/dietpi/cloudshell/scenes/` and/or even something on user basis, like `~/.config/dietpi-cloudshell/scenes`.

But while this would be awesome, if a DietPi-VPN/OpenVPN scene is wanted short-/midterm, a hardcoded additional scene is much easier to implement. A DietPi-VPN scene would be a logical extension, as we have an optional login banner entry for it as well, and some status function in the `dietpi-vpn` script. This could be extended to return some more information for the multiple available lines in a cloudshell scene: https://github.com/MichaIng/DietPi/blob/a27f012/dietpi/func/dietpi-banner#L248

roelbroersma, 2023-11-01:
Exactly, you get the idea!

An openvpn script could also be mader after a split-up of the cloudshell in separate scene files. Let's save us that small part ;)
I think when splitting up the cloudshell, so it would be easier to implement new scenes, will give Cloudshell a serious boost. People will probable add scenes for specific daemons or network-bridging, etc.

If there is anything I could do, let me know.  You probably have a better idea on this, how to separate scenes from the (1 file): https://github.com/MichaIng/DietPi/blob/master/dietpi/dietpi-cloudshell

- Probably, making the aEnabledScenes[] a variable which is loaded from the settings file?
- Create a new array which maps scene_numbers to scene_function_names, e.g:   customScenes[11]="myFunction".
- During the loop of aEnabledScenes, check if there are more than 10 scenes. if so, load the functions files  (.sh) from /etc/dietpi/cloudshell/scenes/   (function files as in the customScenes array).
- Call the function, loaded from the file, by it's name as in the customeScenes array.

In this way,  we first make an extension to cloudShell. So we do not put existing scenes in scene files.  This way it's more seamless to migrate to the new version we create.