MichaIng/DietPi #6710: Radxa ROCK 3A | No sound from 3.5mm jack

wahono77, 2023-10-27:
No second soundcard (jack 3.5mm) in Radxa ROCK 3A

I run :

aplay -l

**** List of PLAYBACK Hardware Devices ****
card 0: HDMI [HDMI], device 0: fe400000.i2s-i2s-hifi i2s-hifi-0 [fe400000.i2s-i2s-hifi i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

My environments:

- DietPi version
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=23
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'

- Distro version : bookworm
- Kernel version : Linux DietPi 6.1.11-rk35xx #23.02.2 SMP PREEMPT Sat Feb 18 00:05:58 UTC 2023 aarch64 GNU/Linux

- SBC model : ROCK 3A (aarch64)
- Power supply used : 5V 1A
- SD card used : SanDisk ultra

MichaIng, 2023-10-28:
Strange indeed. I also see no device tree overlay or something. Btw, we are switching to another (newer) kernel build. In case you have an image backup, you could test it:
```sh
apt install linux-{image,dtb}-current-rockchip64
reboot
# after reboot, if everything works
apt autopurge linux-{image,dtb}-edge-rk35xx
```
The previously dedicated "rk35xx" family builds for RK3566 and RK3568 SBCs have been merged into the generic Rockchip 64-bit kernel builds. The new ones are Linux 6.1.50 and based on a newer Armbian SDK, so potentially contain some fixes or feature additions.

wahono77, 2023-10-29:
Yes, after running your suggestion. HDMI and jack sound 3.5mm is ON. 
Thanks alot @MichaIng

MichaIng, 2023-10-29:
Okay great, many thanks for testing! Then we'll apply this kernel upgrade next DietPi update for every ROCK 3A system.

closed 2023-11-18 by MichaIng