MichaIng/DietPi #6711: Failed to install MineOS

schmidt-software, 2023-10-27:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SD card used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

- ...

Joulinar, 2023-10-27:
Would it be possible to share the error message you are facing?

schmidt-software, 2023-10-27:
Step: Installing MineOS: Minecraft servers with web interface (Java/Node.js)

[  OK  ] DietPi-Software | Checking URL: https://github.com/hexparrot/mineos-node.git
[ INFO ] DietPi-Software | APT install rdiff-backup rsync screen make g++, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
g++ is already the newest version (4:10.2.1-1+rpi1).
make is already the newest version (4.3-4.1).
rdiff-backup is already the newest version (2.0.5-2).
rsync is already the newest version (3.2.3-4+deb11u1).
screen is already the newest version (4.8.0-6).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
[  OK  ] DietPi-Software | APT install rdiff-backup rsync screen make g++
[  OK  ] DietPi-Software | mkdir -p /mnt/dietpi_userdata/mineos
[  OK  ] DietPi-Software | cd /mnt/dietpi_userdata/mineos
[  OK  ] DietPi-Software | cd minecraft
[ INFO ] DietPi-Software | ./update_webui.sh, please wait...
Script execution started on: Fri Oct 27 20:45:29 CEST 2023
Updating mineos-node repository...OK
Pulling master branch...From https://github.com/hexparrot/mineos-node
 * branch            master     -> FETCH_HEAD
OK
Updating npm dependencies.../usr/local/lib/node_modules/npm/lib/es6/validate-engines.js:31
    throw err
    ^

TypeError: LRUCache is not a constructor
    at Object.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/cacache/lib/memoization.js:5:18)
    at Module._compile (node:internal/modules/cjs/loader:1376:14)
    at Module._extensions..js (node:internal/modules/cjs/loader:1435:10)
    at Module.load (node:internal/modules/cjs/loader:1207:32)
    at Module._load (node:internal/modules/cjs/loader:1023:12)
    at Module.require (node:internal/modules/cjs/loader:1235:19)
    at require (node:internal/modules/helpers:176:18)
    at Object.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/cacache/lib/get.js:8:14)
    at Module._compile (node:internal/modules/cjs/loader:1376:14)
    at Module._extensions..js (node:internal/modules/cjs/loader:1435:10)

Node.js v21.1.0
FAILED
[FAILED] DietPi-Software | ./update_webui.sh

Joulinar, 2023-10-28:
hmm for me this is working without issue

```
[ INFO ] DietPi-Software | ./update_webui.sh, please wait...
Script execution started on: Sat Oct 28 19:43:26 CEST 2023
Updating mineos-node repository...OK
Pulling master branch...From https://github.com/hexparrot/mineos-node
 * branch            master     -> FETCH_HEAD
OK
Updating npm dependencies...OK
Setting node javascript files to executable...OK
Script execution ended on: Sat Oct 28 19:43:45 CEST 2023
```

schmidt-software, 2023-10-28:
After reinstalling DietPi on the same USB stick and installing MineOS as the first software package, the installation went smoothly.

closed 2023-10-28 by schmidt-software

MichaIng, 2023-10-28:
Hmm, `./update_webui.sh` is only executed if MineOS was installed before, respectively it the repo was cloned and execute permissions applied to this script. Did it fail before and you retried once?

schmidt-software, 2023-10-29:
Yes, I tried several times.

MichaIng, 2023-10-29:
Okay then it makes sense. Was this an older image with possibly other Node.js software installed? Node sometimes runs into incompatibilities between global (system-wise) and local modules/dependencies. At least we know this from Node-RED. Probably it was similar here.