MichaIng/DietPi #6712: HDMI-CEC not working on Odroid N2+

Ofboir, 2023-10-28:
I would like to be able to control Kodi with my TV remote, but it doesn't work.

#### Required Information
- DietPi version | 8.23.3
- Distro version | Debian Bookworm
- Kernel version | 6.1.50-current-meson64
- SBC model | Odroid N2+ 4 Gb
- Power supply used | 12V 2.08A (Mean Well GST25E12-P1J)
- SD card used | eMMC 128 Gb

#### Software
- Kodi | 2:20.1+dfsg-1
- libcec6 | 6.0.2-5

#### Additional Information (if applicable)
- Result of `cec-client -l` :
`libCEC version: 6.0.2, compiled on Linux ... , features: P8_USB, DRM, P8_detect, randr, Exynos, Linux, AOCEC
Found devices: NONE`

- I tried with the Ubuntu image from Hardkernel (ubuntu-22.04-4.9-mate-odroid-n2-20220629), with the same SBC, same TV and same HDMI cable. Everything works fine. I installed Kodi and it worked instantly. Here is the result of `cec-client -l` :
[image]

MichaIng, 2023-10-28:
We use Armbian's mainline kernel builds, so those cannot be compared to the ancient vendor kernel builds of Hardkernel, and it is expected that not all hardware features are supported (OOTB), especially the rarely used ones. I checked at the Armbian forum and could not find anyone asking for CEC support.

Actually it seems like the kernel driver itself needs to support it. In the kernel docs I see only older Odroid/meson chips being supported: https://www.kernel.org/doc/html/latest/admin-guide/media/cec.html

But it could be also a different issue: https://github.com/home-assistant/operating-system/pull/1717/files

Is the CEC module available, does it load, and in case does it throw errors?
```sh
modinfo cec
modprove cec
dmesg -l 0,1,2,3
```

Ofboir, 2023-10-29:
Hi

Thank you for your answer. I am not familiar at all with the modules mecanisms, so I just pasted blindly the commands you suggested :

`modinfo cec`
> name:           cec
filename:       (builtin)
license:        GPL
file:           drivers/media/cec/core/cec
description:    Device node registration for cec drivers
author:         Hans Verkuil <hans.verkuil@cisco.com>
parm:           debug:debug level (0-2) (int)
parm:           debug_phys_addr:add CEC_CAP_PHYS_ADDR if set (bool)

`modprobe cec`
Nothing on the console

`dmesg -l 0,1,2,3`
> [    1.923562] genirq: Setting trigger mode 3 for irq 23 failed (meson_gpio_irq_set_type+0x0/0x60)
[    4.040467] panfrost ffe40000.gpu: error -ENODEV: _opp_set_regulators: no regulator (mali) found

coder-ant, 2024-01-26:
Same issue here on RPI 2, DietPi v9.0.2

cec-client -l
libCEC version: 6.0.2, compiled on Linux ... , features: P8_USB, DRM, P8_detect, randr, Exynos, Linux, AOCEC
Found devices: NONE

MichaIng, 2024-01-26:
Try with KMS and `cec-ctl`:
```sh
G_CONFIG_INJECT 'dtoverlay=vc4-' 'dtoverlay=vc4-kms-v3d' /boot/config.txt
apt install v4l-utils
reboot
```
See here: https://forums.raspberrypi.com/viewtopic.php?t=348779
Probably `cec-client` works as well with KMS.

coder-ant, 2024-01-28:
Thanks. cec-ctl works fine for me under kms (though kms broke one other command I was using under fkms - vcgencmd display_power, that is a different topic).

MichaIng, 2025-05-31:
Closing this issue. Feel free to reopen if the issue persists.

closed 2025-05-31 by MichaIng