MichaIng/DietPi #6715: Dietpi Update fails - Hyper-V on Windows 11

HyperCriSiS, 2023-10-29:
Hi,

I just tried to update my Dietpi Hyper-V installation from 8.21.1 to 8.23.3.
The update fails.
Bug-report reference code is: 69952347-dfc7-4dc2-8da3-8f2f611b33c9

What can I do?

Thanks
Hyper

Joulinar, 2023-10-29:
Sharing the error message would be a good starting point

HyperCriSiS, 2023-10-29:
I thought this is all in the Bug report.
I managed to fix it with a manual apt upgrade. It installed GRUB which fixed the issue.

closed 2023-10-29 by HyperCriSiS

MichaIng, 2023-10-29:
I guess you needed to manually select the correct bootloader device? For images generated after [this](https://github.com/MichaIng/DietPi/commit/d430094) commit, this should actually not be required anymore, unless you changed something at Hyper-V, like the emulated type of drive, controller or similar, or you added further drives so that `/dev/sda` is not the root drive anymore?

Yes, the bug report upload contains all information. But not everyone has access to them and they life for 48 hours only. So for future reference, and to make it easier for us, also when not being at home with bug report access credentials, it helps to copy&paste the generated GitHub issue template from the console, respectively the related console output.

HyperCriSiS, 2023-10-29:
OK, will do next time :)
Yes, I manually needed to select the proper device. I have an attached USB drive, maybe this is the issue.

MichaIng, 2023-10-29:
> I have an attached USB drive, maybe this is the issue.

Usually internal drives (IDE, SATA, SCSI) should be recognised first and hence become `/dev/sda`. But VMs which are no true hardware may behave different, so that the root drive became `/dev/sdb`. Luckily those device paths are not used by any bootloader or fstab configuration so that this does not cause issues, aside of on GRUB upgrade, when it recognises that there is no GRUB present on the stored target device path.