MichaIng/DietPi #6716: Letsencrypt renewal not updated in Dietpi-Banner

echtfrank, 2023-10-30:
### Creating a bug report/issue

#### Required Information
- DietPi version | G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=23
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'

- Distro version | bookworm
- Kernel version | Linux biblio 6.1.53-current-meson64 #3 SMP PREEMPT Wed Sep 13 07:43:05 UTC 2023 aarch64 GNU/Linux
- SBC model | Odroid C4/HC4 (aarch64)
- Power supply used | 5V 2A
- SD card used | SanDisk ultra 32GB

#### Additional Information (if applicable)
- Software title | Letsencrypt and / or Dietpi-banner
- Was the software title installed freshly or updated/migrated? N/A
- Can this issue be replicated on a fresh installation of DietPi? Unknown / unable to test

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. Start `sudo dietpi-banner`
2. Enable "16  Let's Encrypt cert status"
3. OK, until finished
4. Exit dietpi-banner configuration
5. The banner displays certificate info CORRECT
6. Exit current terminal session
7. Create / login to new terminal session

#### Expected behaviour

- I am logged in and the dietpi-banner displays the Letsencrypt certification information correctly

#### Actual behaviour

- I am logged in and the dietpi-banner displays the Letsencrypt certification, but is is NOT correct
- It displays the text "No certificate found"

#### Extra details

I got a mail notification from Letsencrypt that in a few days my certificate for some (sub) domains will expire.
The main use of my device is hosting a personal Nextcloud instance.  If I check the certifcate of my Nextcloud in a browser, I can see that the certificate has been properly renewed recently and does not expire.

To check the certification expiration, I enabled the respective line in dietpi-banner.
Immediately after applying this setting and leaving dietpi-banner, I am retuned to the bash-prompt and the banner show the correct expiration of the certificate.
But when I exit the terminal session and login to the terminal again, the letsencrypt line is displayed, but does NOT show the correct expiration date but it displays "No certificate found".

This is repeatable: any time I am following the steps in dietpi-banner and return to the terminal prompt, the information is correctly displayed in the banner, but it is NOT after logging into a new terminal-session.

Could it be that there is some wrong historic data in my letsencrypt (configuration) files? The very first time I used Letsencrypt some years ago, I configured it for domain names A, B, C and D. After a a year or so, I reconfigured Letsencrypt for domains A, C, D and E.

Regards,
Frank

MichaIng, 2023-10-31:
Did you login as non-root user? One issue with this banner info is that usually only root has access to the certificate directory `/etc/letsencrypt/live/`. We could show a different info in this case, to not cause confusion. We could also adjust permissions to grant any user access to the contained file lists. Since no one but root has any access to the private keys, this is safe, but I am still a little concerned touching/weakening any of the involved permissions. However, this should be it:
```sh
chmod 0711 /etc/letsencrypt/{archive,live}
```

echtfrank, 2023-11-02:
Thanks, why didn't I think of this myself... Indeed I login as non-root user.
And thanks for the given solution; I share your concern for weakening the permissions, so I won't do that either.

I am going to try to manage a better aproach using a user defined entry. Appreciated your help.