MichaIng/DietPi #6718: Asus tinker board 2 doesn't boot

onurcoskun14, 2023-10-31:
### Creating a bug report/issue

#### Required Information
- DietPi version | latest
- Distro version | debiab bookworm
- Kernel version | idk
- SBC model | Asus tinker board 2
- Power supply used | 12v 5a
- SD card used |  SanDisk ultra 16gb a1 HCI

#### Steps to reproduce
1. Download latest Asus tinker board dietpi
2. Write to sd card With  BalenaEtcher
3. Put sd card in and plug it into power.
4. Watch screen

#### Expected behaviour
Should see something on screen
#### Actual behaviour
system  does nothing no signal on hdmi.

MichaIng, 2023-10-31:
Since ASUS Tinker Board 2 support and images have not yet been announced, which image did you flash? There is none on our download page yet. However, I need someone to test our testing images anyway, so if this was not the case anyway, here you can find them: https://dietpi.com/downloads/images/testing/

onurcoskun14, 2023-10-31:
Sorry about that i was trying to boot Asus tinker board image. I can try test images on my Asus TB 2.

MichaIng, 2023-10-31:
Then it makes sense. The ASUS Tinker Board (1) is a very different SBC with older 32-bit SoC while ATB 2 uses a newer 64-bit SoC. Those are incompatible. Let me know if the new ones work, then I'll add them to our download page and release announcements as well 🙂.

onurcoskun14, 2023-11-01:
I will try today and let you know.

onurcoskun14, 2023-11-01:
I successfully boot the asus tb2 but there is led blinking orange nonstop(not the power led). And after first installation i didn't setup any software but NetworkManager. But the problem is i cant scan anything.

Joulinar, 2023-11-01:
`NetworkManager` is a conflicting application for `ifupdown` we use to manage network connection.

onurcoskun14, 2023-11-01:
Thank you for ur answer i manged to solve the issue. But there is a problem with reboot. When i reboot the system is takes a lot of time and the reason is 
job ifup@eth0.service/start, job ifup@wlan0.service/start  running is the message. It takes about 1 min on every boot

Joulinar, 2023-11-01:
You are using DHCP? If yes, how does it behave using STATIC IP?

MichaIng, 2023-11-01:
Also it seems you have WiFi and Ethernet enabled concurrently? At least on first boot, only one of both is enabled, and the way `dietpi-config` sets up `/etc/network/interfaces` does not really support two concurrent network interfaces (unless you install the WiFi hotspot), as both are setup with a default route/gateway, conflicting this way.

If you want to use NetworkManager, at best remove the content of `/etc/network/interfaces` to assure `ifupdown` is not interfering.

onurcoskun14, 2023-11-01:
> Also it seems you have WiFi and Ethernet enabled concurrently? At least on first boot, only one of both is enabled, and the way `dietpi-config` sets up `/etc/network/interfaces` does not really [...]

Thank you this worked like a charm.

MichaIng, 2023-11-01:
Okay, aside of the networking tool conflicts (unrelated to the SBC/kernel etc), our ASUS Tinker Board 2 image runs fine? I'd add it to our download page, docs etc and add an announcement to next release notes in this case.

onurcoskun14, 2023-11-01:
I haven't encountered any problems so far.

MichaIng, 2023-11-01:
Many thanks. I'll mark this issue as closed then. Feel free to reopen if you face any issues.

closed 2023-11-01 by MichaIng

onurcoskun14, 2023-11-09:
Hello again,
I have been using my ATB 2 with dietpi for a while. I have encountered with 3 problems.
1. Status led 3 is keep blinking orange nonstop.
2. I could not use serial communication via pins
3. System stoped recognizing my Bluetooth Dongle

MichaIng, 2023-11-10:
> Status led 3 is keep blinking orange nonstop.

Can you change its behaviour in `dietpi-led_control`?

> I could not use serial communication via pins

Which UART are you trying to use in particular? Please try UART0 at pins 6, 8 and 10.

> System stoped recognizing my Bluetooth Dongle after some time.

But it did work before, including actual connection to Bluetooth devices? Please check for kernel errors:
```sh
dmesg -l 0,1,2,3
```

onurcoskun14, 2023-11-14:
1. Was not a problem my fault.
2. I am using pins 6, 8 and 10 and trying to comm via ttyS0 this the output
[image]
in /dev/tty there are ttyS0 to ttyS7 i tried all of them only ttyS2 did not give any exception but still not working beacuse it does not  sent data to my other device.
3. About bluetooth onboard bluetooth adapter is not showing in bluetoothctl. Now i can see my dongle when i plug it to ATB2.
this is the out of **dmesg -l 0,1,2,3**.
[image]

MichaIng, 2023-11-16:
> in /dev/tty there are ttyS0 to ttyS7 i tried all of them only ttyS2 did not give any exception but still not working beacuse it does not sent data to my other device.

It is expected that only `ttyS2` is up by default. There is a device tree overlay to enable "UART4", but not sure which device and pins this is exactly. In the readme they are labelled "GPIO1_A7" and "GPIO1_B0".

Note that boot output is done at 1.5M baud rate, so with the default 115200 there is likely one a few scrambled characters on the UART client.

> About bluetooth onboard bluetooth adapter is not showing in bluetoothctl. Now i can see my dongle when i plug it to ATB2.
this is the out of dmesg -l 0,1,2,3.

I guess "hci0" is the onboard Bluetooth adapter? It seems to not find a matching firmware 🤔. Unexpected with the specific Armbian firmware package. ... indeed it is not there:

[image]

Checking `armbian-firmware-full`:

[image]

There it is. Please install that package and check whether Bluetooth works then:
```sh
apt install armbian-firmware-full
```

onurcoskun14, 2023-11-20:
1. Issue about onboard bluetooth adapter is resolved after ur instructions.
2. Issue about external bluetooth dongle i thought i was resolved but i was not. I have tried atleast 5 different bluetooth dongle they system sometime recognize the dongle but after a reboot then it does not. There is something going on that i dont understand. When it doesnt recognize the dongle it is not shown in **lsusb**.
3. About serial communication There is a device tree overlay to enable "UART4" do you mean from here
[image]
if you mean here i dont wanna use it as console. If not i dont know where to enable it. I have tried using  "GPIO1_A7" and "GPIO1_B0" this but still didnt work.

MichaIng, 2023-12-22:
Hmm, if the raw USB device is not detected, then it is at least no issue with driver or firmware. Those "device descriptor read" errors look more like an issue with the USB port/adapter, powering or so. It is known issue with some RK3399 boards the the USB3 ports do not work, or not well, with the current kernel. Did you try different USB ports, especially USB2 ones?

With "overlay" I mean a kernel device tree overlay. You can enable it like this:
```sh
G_CONFIG_INJECT 'overlays=' 'overlays=uart4' /boot/dietpiEnv.txt
```
Reboot required.

The `dietpi-config` menu you found is for enabling a login console on a serial device. It just allows this for the device nodes found below `/dev/`, without knowing which ones are actually functional (for this purpose) or not.