MichaIng/DietPi #6719: SSH first login - "DietPi first run setup is currently running on another screen" undefinitely

inis17, 2023-10-31:
### Creating a bug report/issue
https://github.com/MichaIng/DietPi/issues/1443
#### Required Information
- DietPi version 
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=23
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'

- Distro version
bookworm 0

- Kernel version
Linux DietRPI 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

- SBC model 
RPi 4 Model B (aarch64)

- Power supply used 
5V 1A RAVpower
- SD card used | SSD Samsung EVO 870 1TO

#### Additional Information (if applicable)
- Can this issue be replicated on a fresh installation of DietPi?
This happened on a fresh flash of the SSD with the latest dietpi image : DietPi_RPi-ARMv8-Bookworm
this is the dietpi.txt used :
```bash
AUTO_SETUP_LOCALE=C.UTF-8
AUTO_SETUP_KEYBOARD_LAYOUT=fr
AUTO_SETUP_TIMEZONE=Europe/Paris
AUTO_SETUP_NET_ETHERNET_ENABLED=1

AUTO_SETUP_NET_WIFI_ENABLED=0
AUTO_SETUP_NET_WIFI_COUNTRY_CODE=GB

AUTO_SETUP_NET_USESTATIC=0
AUTO_SETUP_NET_STATIC_IP=192.168.0.100
AUTO_SETUP_NET_STATIC_MASK=255.255.255.0
AUTO_SETUP_NET_STATIC_GATEWAY=192.168.0.1
AUTO_SETUP_NET_STATIC_DNS=9.9.9.9 149.112.112.112
AUTO_SETUP_DHCP_TO_STATIC=0

AUTO_SETUP_NET_HOSTNAME=DietRPI
AUTO_SETUP_NET_ETH_FORCE_SPEED=0
AUTO_SETUP_BOOT_WAIT_FOR_NETWORK=1
AUTO_SETUP_SWAPFILE_SIZE=1
AUTO_SETUP_SWAPFILE_LOCATION=/var/swap
AUTO_SETUP_HEADLESS=1
AUTO_UNMASK_LOGIND=0

AUTO_SETUP_CUSTOM_SCRIPT_EXEC=0
AUTO_SETUP_BACKUP_RESTORE=0

AUTO_SETUP_SSH_SERVER_INDEX=-2
AUTO_SETUP_SSH_PUBKEY=ssh-rsa exampleKey

AUTO_SETUP_LOGGING_INDEX=-2
AUTO_SETUP_RAMLOG_MAXSIZE=50

AUTO_SETUP_WEB_SERVER_INDEX=0
AUTO_SETUP_DESKTOP_INDEX=0
AUTO_SETUP_BROWSER_INDEX=0
AUTO_SETUP_AUTOSTART_TARGET_INDEX=0
AUTO_SETUP_AUTOSTART_LOGIN_USER=root
AUTO_SETUP_AUTOMATED=1 
AUTO_SETUP_INSTALL_SOFTWARE_ID=134 #Docker Compose
AUTO_SETUP_INSTALL_SOFTWARE_ID=162 #Docker

AUTO_SETUP_GLOBAL_PASSWORD=example123
SURVEY_OPTED_IN=0

CONFIG_CPU_GOVERNOR=schedutil
CONFIG_CPU_ONDEMAND_SAMPLE_RATE=25000
CONFIG_CPU_ONDEMAND_SAMPLE_DOWNFACTOR=40
CONFIG_CPU_USAGE_THROTTLE_UP=50
CONFIG_CPU_MAX_FREQ=Disabled
CONFIG_CPU_MIN_FREQ=Disabled
CONFIG_CPU_DISABLE_TURBO=0
CONFIG_GPU_DRIVER=none

CONFIG_PROXY_ADDRESS=MyProxyServer.com
CONFIG_PROXY_PORT=8080
CONFIG_PROXY_USERNAME=
CONFIG_PROXY_PASSWORD=
CONFIG_G_CHECK_URL_TIMEOUT=10
CONFIG_G_CHECK_URL_ATTEMPTS=2

CONFIG_CHECK_CONNECTION_IP=9.9.9.9
CONFIG_CHECK_CONNECTION_IPV6=2620:fe::fe
CONFIG_CHECK_DNS_DOMAIN=dns9.quad9.net

CONFIG_CHECK_DIETPI_UPDATES=1
CONFIG_CHECK_APT_UPDATES=2

CONFIG_NTP_MODE=2

CONFIG_SERIAL_CONSOLE_ENABLE=0

CONFIG_SOUNDCARD=none
CONFIG_LCDPANEL=none

CONFIG_ENABLE_IPV6=0

CONFIG_APT_RASPBIAN_MIRROR=http://raspbian.raspberrypi.org/raspbian/
CONFIG_APT_DEBIAN_MIRROR=https://deb.debian.org/debian/
CONFIG_NTP_MIRROR=debian.pool.ntp.org

SOFTWARE_DISABLE_SSH_PASSWORD_LOGINS=root

SOFTWARE_VNCSERVER_WIDTH=1280
SOFTWARE_VNCSERVER_HEIGHT=720
SOFTWARE_VNCSERVER_DEPTH=16
SOFTWARE_VNCSERVER_DISPLAY_INDEX=1
SOFTWARE_VNCSERVER_SHARE_DESKTOP=0
SOFTWARE_OWNCLOUD_NEXTCLOUD_USERNAME=admin
SOFTWARE_OWNCLOUD_DATADIR=/mnt/dietpi_userdata/owncloud_data
SOFTWARE_NEXTCLOUD_DATADIR=/mnt/dietpi_userdata/nextcloud_data
SOFTWARE_WIFI_HOTSPOT_SSID=DietPi-HotSpot
SOFTWARE_WIFI_HOTSPOT_KEY=dietpihotspot
SOFTWARE_WIFI_HOTSPOT_CHANNEL=3
SOFTWARE_WIFI_HOTSPOT_WIFI4=0
SOFTWARE_WIFI_HOTSPOT_WIFI5=0
SOFTWARE_WIFI_HOTSPOT_WIFI6=0
SOFTWARE_WIFI_HOTSPOT_5G=0
SOFTWARE_WIFI_HOTSPOT_5G_CHANNEL=36
SOFTWARE_XORG_DPI=96
SOFTWARE_CHROMIUM_RES_X=1280
SOFTWARE_CHROMIUM_RES_Y=720
SOFTWARE_CHROMIUM_AUTOSTART_URL=https://dietpi.com/
SOFTWARE_HOMEASSISTANT_APT_DEPS=
SOFTWARE_HOMEASSISTANT_PIP_DEPS=
SOFTWARE_K3S_EXEC=
SOFTWARE_DIETPI_DASHBOARD_VERSION=Stable
SOFTWARE_DIETPI_DASHBOARD_BACKEND=0
SOFTWARE_URBACKUP_BACKUPPATH=/mnt/dietpi_userdata/urbackup
DEV_GITBRANCH=master
DEV_GITOWNER=MichaIng
```

#### Steps to reproduce
1. Just flash the image and use the previous dietpi.txt config file and the ssh session gets stuck into : 
[image]

#### Expected behaviour
The first boot should complete, install openSSH, Docker, Docker compose

#### Actual behaviour
nothing beside the screeshot 

#### Extra details
https://github.com/MichaIng/DietPi/issues/1443
this thread helped solve the issue and the command stop the auto setup
```bash
rm /tmp/.dietpi-login_firstrun_setup.pid
rm -R  /etc/systemd/system/getty@.service.d
/boot/dietpi/dietpi-login
```

MichaIng, 2023-10-31:
Instead of potentially running two concurrent first run setup processes, better check the status of the original one via `htop`, or attach a screen to watch it. It can take quite a while to finish.

inis17, 2023-10-31:
i reproduced my steps, 
[image]

[image]

MichaIng, 2023-10-31:
Looks all as expected: `dietpi-software` is running on the main screen/TTY. You see the log file `/var/tmp/dietpi/logs/dietpi-firstrun-setup.log`:
```sh
tail -f /var/tmp/dietpi/logs/dietpi-firstrun-setup.log
```
Wait for it to finish, then the login message, which tells you exactly what happens, should disappear.

inis17, 2023-10-31:
this is the result of `tail -f /var/tmp/dietpi/logs/dietpi-firstrun-setup.log`

```bash
[ INFO ] DietPi-Software | DietPi has not fully been installed.
This must be completed prior to using DietPi by selecting:
 - Go Start Install.
```

MichaIng, 2023-10-31:
Hmm, this message should not show up if automated setup was enabled. Also it indicates that it is not non-interactive (anymore). Can you show the whole log file?
```sh
cat /var/tmp/dietpi/logs/dietpi-firstrun-setup.log
```
And additionally:
```sh
echo "'$(sed -n '/^[[:blank:]]*AUTO_SETUP_AUTOMATED=/{s/^[^=]*=//p;q}' /boot/dietpi.txt)'"
```

MichaIng, 2023-10-31:
Fixed with: https://github.com/MichaIng/DietPi/commit/d1ebbbb

closed 2023-11-18 by MichaIng