MichaIng/DietPi #6720: Failed to install Docker-CE on a Fresh Flashed SSD - No Space left on device

inis17, 2023-11-01:
#### Details:
- Date           | Wed Nov  1 10:34:01 CET 2023
- Bug report     | 01069e9a-86e8-41ed-a692-39635b3933e8
- DietPi version | v8.23.3 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux DietRPI 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bookworm (ID=7,RASPBIAN=0)
- Command        | `apt-get -y --allow-change-held-packages install docker-ce`
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:
on a Fresh install of DietPi on the SSD - I run `dietpi-software` to install `docker` and `docker compose`
#### Expected behaviour:

Install process should suceed
#### Actual behaviour:

Install process fails and the i have a `failed to write (No space left on device`
[image]

#### Extra details:
i have this problem only since i update the image to bookworm.

Joulinar, 2023-11-01:
your file system seems to be full, Can you share

```
df -h
```

inis17, 2023-11-01:
```bash
root@DietRPI:~# df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       700M  666M     0 100% /
devtmpfs        3.9G     0  3.9G   0% /dev
tmpfs           3.9G     0  3.9G   0% /dev/shm
tmpfs           1.6G  8.6M  1.6G   1% /run
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           3.9G     0  3.9G   0% /tmp
tmpfs            50M     0   50M   0% /var/log
/dev/sda1       127M   35M   93M  28% /boot
```

Joulinar, 2023-11-01:
> ```
> Filesystem      Size  Used Avail Use% Mounted on
> /dev/root       700M  666M     0 100% /
> ```

how large is your SSD? Current size seems to be below 1GB. Can you share following?

```
lsblk -o name,fstype,label,size,ro,type,mountpoint,partuuid,uuid
```

Are there any kernel error messages?

```
dmesg -l 0,1,2,3
```

inis17, 2023-11-01:
```bash 
root@DietRPI:~# lsblk -o name,fstype,label,size,ro,type,mountpoint,partuuid,uuid
NAME FSTYPE LABEL   SIZE RO TYPE MOUNTPOINT PARTUUID                             UUID
sda               931.5G  0 disk
├─sda1
│    vfat           128M  0 part /boot      c06c5cdf-01                          74EC-57EC
└─sda2
     ext4         931.4G  0 part /          c06c5cdf-02                          90a1c2df-598c-4b9b-bdde-5a1afc09bb28
```
```bash
root@DietRPI:~# dmesg -l 0,1,2,3
[    0.626092] bcm2708_fb soc:fb: Unable to determine number of FBs. Disabling driver.
[    0.811687] mmc1: Controller never released inhibit bit(s).
```

Joulinar, 2023-11-01:
hmm strange, `/dev/root` is shown with 700MB only, while the whole partition `sda2` has 931GB. 

@MichaIng any idea how this could happen?

inis17, 2023-11-01:
i used belenaEtcher on Macos Sonoma 14.0 to flash the SSD with the DietPi_RPi-ARMv8-Bookworm.img.xz
(if this is any help)

Joulinar, 2023-11-01:
was this a fresh install? Probably the partition did not get extended. If this was a fresh install, can you share logs from `/var/tmp/dietpi/logs/` You can attach the files to GitHub 😉

inis17, 2023-11-01:
[dietpi-firstboot.log](https://github.com/MichaIng/DietPi/files/13226397/dietpi-firstboot.log)
[dietpi-firstrun-setup.log](https://github.com/MichaIng/DietPi/files/13226398/dietpi-firstrun-setup.log)
[dietpi-ramlog.log](https://github.com/MichaIng/DietPi/files/13226399/dietpi-ramlog.log)
[fs_partition_resize.log](https://github.com/MichaIng/DietPi/files/13226400/fs_partition_resize.log)

Joulinar, 2023-11-01:
already first run setup failed. You should have seen the error before already. Did you?

```
dpkg: error processing archive /tmp/apt-dpkg-install-3eIffr/12-rpi-eeprom_20.0-2_all.deb (--unpack):
 cannot copy extracted data for './lib/firmware/raspberrypi/bootloader-2711/latest/pieeprom-2023-05-11.bin' to '/lib/firmware/raspberrypi/bootloader-2711/latest/pieeprom-2023-05-11.bin.dpkg-new': failed to write (No space left on device)
Errors were encountered while processing:
 /tmp/apt-dpkg-install-3eIffr/12-rpi-eeprom_20.0-2_all.deb
```

Partition seems to have extended

```
/dev/sda2: 
New situation:
Disklabel type: dos
Disk identifier: 0xc06c5cdf

Device     Boot  Start        End    Sectors   Size Id Type
/dev/sda1  *      2048     264191     262144   128M  c W95 FAT32 (LBA)
/dev/sda2       264192 1953525167 1953260976 931.4G 83 Linux

The partition table has been altered.
```

But somehow it's not the case. 🤔

inis17, 2023-11-01:
yes i have seen the eeprom install failed maybe it is because i had to update the rpi-eeprom with a SD card of raspeberry OS in order to change the boot priority ? (but long prior)

Joulinar, 2023-11-01:
> because i had to update the rpi-eeprom with a SD card of raspeberry OS in order to change the boot priority 

This has nothing to do with it.

inis17, 2023-11-01:
```bash
root@DietRPI:/var/tmp/dietpi# resize2fs "$G_ROOTFS_DEV"
resize2fs 1.47.0 (5-Feb-2023)
Filesystem at /dev/sda2 is mounted on /; on-line resizing required
old_desc_blocks = 1, new_desc_blocks = 117
The filesystem on /dev/sda2 is now 244157622 (4k) blocks long.
```
it seems to have fixed my issue, thank you for your help !

Joulinar, 2023-11-01:
well first run did not finished and there might be packages missing now

MichaIng, 2023-11-01:
Indeed, run `dietpi-software` to finish it.

inis17, 2023-11-02:
After more investigation : 
I had an error during the first instal every time with rpi-eeprom
```bash
dpkg: error processing archive /tmp/apt-dpkg-install-3eIffr/12-rpi-eeprom_20.0-2_all.deb (--unpack):
 cannot copy extracted data for './lib/firmware/raspberrypi/bootloader-2711/latest/pieeprom-2023-05-11.bin' to '/lib/firmware/raspberrypi/bootloader-2711/latest/pieeprom-2023-05-11.bin.dpkg-new': failed to write (No space left on device)
Errors were encountered while processing:
 /tmp/apt-dpkg-install-3eIffr/12-rpi-eeprom_20.0-2_all.deb
```
Turns out updating it using a SD card flashed for it solved this issue and i c

closed 2023-11-02 by inis17

Joulinar, 2023-11-02:
The issue is not the package. It's the missed extension of your partition. Therefore you don't have enough space to install or update packages.

MichaIng, 2023-11-05:
Probably UAS, respectively that the SSD does not support it well, was the underlying reason why the filesystem expansion failed, or why even that it did succeed now (as of above comment) it still somehow does not provide this additional space. Not sure about the range of issues UAS can cause if its not supported.

Lately issues regarding UAS are rare, but for reference: #3754