MichaIng/DietPi #6722: NFS-Server Install Fail

supertevran, 2023-11-01:
### Details:
- Date           | Wed Nov  1 20:18:21 CET 2023
- DietPi version | v8.23.3 (MichaIng/master)
- Image creator  | 
- Pre-image      | 
- Hardware       | RPi 3 Model B (aarch64) (ID=3)
- Kernel version | `Linux DietPi3B 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bookworm (ID=7,RASPBIAN=0)
- Command        | `echo '/mnt/dietpi_userdata *(rw,async,no_root_squash,fsid=0,crossmnt,no_subtree_check)' > /etc/exports.d/dietpi.exports`
- Exit code      | 127
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
/boot/dietpi/func/dietpi-globals: line 853: echo '/mnt/dietpi_userdata *(rw,async,no_root_squash,fsid=0,crossmnt,no_subtree_check)' > /etc/exports.d/dietpi.exports: No such file or directory
```

supertevran, 2023-11-02:
Hi. Ran the command successfully. After 2nd installation evrything was fine and the NFS-Server came up fine. Thx. a lot for your help!

Joulinar, 2023-11-02:
Let me reopen as we need to find out why this happens

MichaIng, 2023-11-02:
Fixed with: https://github.com/MichaIng/DietPi/commit/06bea90
Changelog: https://github.com/MichaIng/DietPi/commit/7e7ce4f

PR for a live patch for current DietPi version: #6723

closed 2023-11-02 by MichaIng