MichaIng/DietPi #6725: Dietpi-Software Adguard not installing.

lutfor-diu, 2023-11-03:
#### Details:
- Date           | Fri Nov  3 22:32:01 +06 2023
- Bug report     | 7b4f5658-243b-40a8-a259-34f34de73f97
- DietPi version | v8.23.3 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       | RPi B (armv6l) (ID=1)
- Kernel version | `Linux DietPi 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux`
- Distro         | bookworm (ID=7,RASPBIAN=1)
- Command        | `tar xf AdGuardHome_linux_armv6.tar.gz`
- Exit code      | 2
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```

```

Joulinar, 2023-11-03:
if you are hit by the issue, can you scroll up a bit on your SSH terminal to check last lines before the error screen?

lutfor-diu, 2023-11-04:
I think it was my ISP issue.  No problem when try to reinstall after few hours.

Thanks a lot.

closed 2023-11-04 by lutfor-diu