MichaIng/DietPi #6733: `apt-get -y -eany update`

Macampur, 2023-11-07:
#### Details:
- Date           | Tue Nov  7 04:19:25 GMT 2023
- Bug report     | 004342fd-1816-4bc5-a9c4-a5a533a07293
- DietPi version | v8.23.3 (MichaIng/master)
- Image creator  | 
- Pre-image      | 
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bookworm (ID=7,RASPBIAN=0)
- Command        | `apt-get -y -eany update`
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Hit:1 https://deb.debian.org/debian bookworm InRelease
```

Joulinar, 2023-11-07:
can you share some more details

```
apt update
```

MichaIng, 2023-11-07:
From the bug report upload:
```
[ 1362.276138] hwmon hwmon1: Undervoltage detected!
[ 1378.404136] hwmon hwmon1: Voltage normalised
[ 1380.420124] hwmon hwmon1: Undervoltage detected!
[ 1388.484187] hwmon hwmon1: Voltage normalised
[ 1392.516145] hwmon hwmon1: Undervoltage detected!
[ 1452.996529] hwmon hwmon1: Voltage normalised
[ 1457.028230] hwmon hwmon1: Undervoltage detected!
[ 1473.156161] hwmon hwmon1: Voltage normalised
[ 1475.172138] hwmon hwmon1: Undervoltage detected!
[ 1483.236185] hwmon hwmon1: Voltage normalised
[ 1489.284164] hwmon hwmon1: Undervoltage detected!
[ 1523.556144] hwmon hwmon1: Voltage normalised
[ 1525.572119] hwmon hwmon1: Undervoltage detected!
```
You need to use a better PSU. Might or might not be the reason for this sudden aborted APT update as well. Otherwise no error or related log visible, so above command should reveal it (or it was some temporary issue only).

adrian-vN, 2023-11-17:
Having the same problem today - tried updating from 8.22.3 to 8.23.3

#### Details:
- Date           | Fri Nov 17 17:40:00 ACDT 2023
- DietPi version | v8.22.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | `Linux DietPi 5.10.103-v7+ #1529 SMP Tue Mar 8 12:21:37 GMT 2022 armv7l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `apt-get -y -eany update`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...  Run dietpi-update...
#### Expected behaviour:

- ...Updater runs and completes successfully - upgrade to 8.23.3
#### Actual behaviour:

- ...Error as described and update does not complete.  Stays on current version 8.22.3
#### Extra details:

- ...Also had an NTP error but resolved by running apt -y install systemd-timesyncd
#### Additional logs:
```
Hit:1 https://download.docker.com/linux/raspbian bullseye InRelease
```

adrian-vN, 2023-11-17:
root@DietPi:~# apt update
Get:1 http://raspbian.mirror.digitalpacific.com.au/raspbian bullseye InRelease [15.0 kB]
Hit:2 https://download.docker.com/linux/raspbian bullseye InRelease                                                                                                   
Get:3 https://pkgs.tailscale.com/stable/raspbian bullseye InRelease                                                                 
Hit:4 https://archive.raspberrypi.org/debian bullseye InRelease                          
Hit:5 http://phoscon.de/apt/deconz bullseye-beta InRelease
Fetched 21.5 kB in 4s (5,336 B/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.

Joulinar, 2023-11-17:
That all looks good

MichaIng, 2023-11-17:
Jep, please retry software installation. Probably it was a temporary network or APT mirror issue.

_I need to have a look into our code why only a single line is printed for the bug report template 🤔._

adrian-vN, 2023-11-17:
Well, there you go - it worked!  Ha :)  Thanks

MichaIng, 2023-11-17:
Great, I'll mark the issue as closed then 🙂.

closed 2023-11-17 by MichaIng

Macampur, 2023-11-18:
I solved the problem by deleting the repository: **pkg.ltec.ch bookworm** following the instructions in the ticket: **https://dietpi.com/forum/t/dietpi-update-error-with-release-file-pkg-ltec-ch-bookworm /17519**