MichaIng/DietPi #6736: DietPi-Software | Update Vaultwarden to v1.30.0

pdsakurai, 2023-11-07:
Hi!

May I request to update the binaries of Vaultwarden in your server?

Vaultwarden has a new release v1.30.0. It contains the fix for crashes that happen on mobile apps.

[Here](https://github.com/dani-garcia/vaultwarden/releases/tag/1.30.0)'s the link to its release notes.

Thanks!

Regards,
Paul

Joulinar, 2023-11-08:
A new build has been started. https://github.com/MichaIng/DietPi/actions/runs/6795652298
It will take a couple of hours to complete.

MichaIng, 2023-11-08:
To test the new builds:
```sh
cd /tmp
curl -fo package.deb "https://dietpi.com/downloads/binaries/$G_DISTRO_NAME/testing/vaultwarden_$G_HW_ARCH_NAME.deb"
dpkg -i package.deb
rm package.deb
```

MichaIng, 2023-11-08:
New packages have been installed on the server.

pdsakurai, 2023-11-08:
Confirmed using the command `dietpi-software reinstall 183`.

closed 2023-11-08 by pdsakurai