MichaIng/DietPi #6738: Can not install OpenVPN

onurcoskun14, 2023-11-08:
### Creating a bug report/issue

#### Required Information
- DietPi version | 8
- Distro version | bookworm
- Kernel version | Linux DietPi 6.1.21-v8+ #1642 
- SBC model | Rpi 4
- Power supply used | 5v 3a
- SD card used | SanDisk ultra

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | 74e387cb-e091-4c9c-b10b-89170b43e8f8

#### Steps to reproduce

1. sudo apt-get install openvpn

#### Expected behaviour

- should install openvpn

#### Actual behaviour

- lzma_read: Read error(10)

#### Extra details

- ...

Joulinar, 2023-11-08:
can you share whole output of

onurcoskun14, 2023-11-08:
[image]

MichaIng, 2023-11-08:
Looks like the xz-compressed lists (or one of them) is broken, or the extractor. Try this:
```sh
/boot/dietpi/func/dietpi-set_software apt clean
apt update
```

onurcoskun14, 2023-11-08:
Worked thanks.

closed 2023-11-08 by MichaIng