MichaIng/DietPi #6739: Unable to install Docker from DietPi Software or APT

boterocamilo, 2023-11-08:
#### Details:
- Date           | Wed Nov  8 13:31:59 GMT 2023
- Bug report     | bcd872a6-1671-4cf7-a1f1-5adb3c63a7ed
- DietPi version | v8.23.3 (MichaIng/master)
- Image creator  | 
- Pre-image      | 
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bookworm (ID=7,RASPBIAN=0)
- Command        | `apt-get -y -eany update`
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. dietpi-software
2. Search/Browse software
3. Select Docker (both Docker compose and Docker)
4. Select install
5. Error found without the possibility to install Docker.
#### Expected behaviour:

- Docker needs to be installed.
#### Actual behaviour:

- Docker is not installed.
#### Extra details:

- This is the entire result after running DietPi-Software:
[  OK  ] DietPi-Software | Initialised database
[  OK  ] DietPi-Software | Reading database
[  OK  ] DietPi-Software | DietPi-Userdata validation: /mnt/dietpi_userdata

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Checking for conflicts and missing inputs

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Checking for prerequisite software

[  OK  ] DietPi-Software | mkdir -p /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads /var/www /opt /usr/local/bin
[  OK  ] DietPi-Software | chown dietpi:dietpi /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads
[  OK  ] DietPi-Software | chmod 0775 /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads
[ SUB1 ] DietPi-Services > stop 
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : radarr
[  OK  ] DietPi-Services | stop : sonarr
[  OK  ] DietPi-Services | stop : jackett
[  OK  ] DietPi-Services | stop : jellyfin
[  OK  ] DietPi-Services | stop : qbittorrent
[ INFO ] DietPi-Software | APT update, please wait...
Hit:1 https://deb.debian.org/debian bookworm InRelease
Hit:2 https://deb.debian.org/debian bookworm-updates InRelease
Hit:3 https://download.docker.com/linux/debian bookworm InRelease
Hit:4 https://deb.debian.org/debian-security bookworm-security InRelease
Hit:5 https://apt.sonarr.tv/debian buster InRelease
Hit:6 https://deb.debian.org/debian bookworm-backports InRelease
Hit:7 https://download.mono-project.com/repo/debian buster InRelease
Hit:8 https://archive.raspberrypi.org/debian bookworm InRelease
Ign:9 https://download.docker.com/linux/ubuntu bookworm InRelease
Err:10 https://download.docker.com/linux/ubuntu bookworm Release
  _**404  Not Found [IP: 18.173.5.22 443]**_
Hit:11 https://repo.jellyfin.org/debian bookworm InRelease
Reading package lists...
E: The repository 'https://download.docker.com/linux/ubuntu bookworm Release' does not have a Release file.
W: https://download.docker.com/linux/debian/dists/bookworm/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
[FAILED] DietPi-Software | APT update
 - Command: apt-get -y -eany update
[ INFO ] DietPi-BugReport | Generating informative command outputs, please wait...
[ INFO ] DietPi-BugReport | cp /tmp/G_EXEC_ERROR_REPORT G_EXEC_ERROR_REPORT, please wait...
[  OK  ] DietPi-BugReport | cp /tmp/G_EXEC_ERROR_REPORT G_EXEC_ERROR_REPORT
[  OK  ] DietPi-BugReport | Packing upload archive
[  OK  ] DietPi-BugReport | Sending bug report
[  OK  ] DietPi-BugReport | Your bug report has been successfully uploaded.
- Reference code: bcd872a6-1671-4cf7-a1f1-5adb3c63a7ed
#### Additional logs:
```
Hit:1 https://deb.debian.org/debian bookworm InRelease
```
---------------------------------------------------------------------
[FAILED] DietPi-Software | Unable to continue, DietPi-Software will now terminate.

MichaIng, 2023-11-08:
You have a conflicting Ubuntu Docker repository `https://download.docker.com/linux/ubuntu`, which . Check your list files in `/etc/apt/sources.list.d` and remove the one which contains this string.

boterocamilo, 2023-11-08:
I have tried first with these two:
- https://docs.docker.com/engine/install/ubuntu/ (Maybe coming from here the problem because Ubuntu)
- https://www.fosslinux.com/49959/install-docker-on-debian.htm

Joulinar, 2023-11-08:
ok closing this as it was out of our scripting.

closed 2023-11-08 by Joulinar

MichaIng, 2023-11-08:
Jep, DietPi is Debian, so this would have worked: https://docs.docker.com/engine/install/debian/
Take care to cleanup leftovers from other guides/scripts, otherwise you have multiple list files, APT keys etc which may conflict. They are all doing essentially the same, installing a key and a list file, but the key location and files names are different.