MichaIng/DietPi #6740: Rock 3A no DSI display output and wrong configuration for HDMI display out

KSuondProject, 2023-11-10:
No DSI signal and power for 5" DSI dispaly 800x480 [https://www.farnell.com/datasheets/3708790.pdf](url)
Wrong signal for HDMI output for other 5" HDMI display 800x480 [https://www.farnell.com/datasheets/2884058.pdf](url)

- DietPi version | G_DIETPI_VERSION_CORE=8
                           G_DIETPI_VERSION_SUB=23
                           G_DIETPI_VERSION_RC=3
                           G_GITBRANCH='master'
                           G_GITOWNER='MichaIng'
                           G_LIVE_PATCH_STATUS[0]='applied'
                           G_LIVE_PATCH_STATUS[1]='applied'

- Kernel version | Linux/arm64 6.1.50 Kernel

- SBC model | RAXDA ROCK 3A  4 GB

- Power supply used |  5V 5A 

- SD card used | SanDisk ultra)

#### Additional Information (if applicable)
Tried many power supply

#### Steps to reproduce
Installed the latest dietpi img and connected correctly trough SSH.
the display with DSI connection is attached with the flat cable but there's no power (led doesn't light up) e no video signal.
Attaching the other HDMI display the video is shown but at the wrong resolution and it is splice in the half of the display.
using dietpi-config -> display settings -> the system says "no display available for this architecture"

#### Expected behaviour
Work booth display correctly

#### Extra details
There's some overlay to apply? If yes can someone help me to find the correct and explain how to use with dietpi?

Thanks in advance.

KSuondProject, 2023-11-20:
No way to make it work correctly?

KSuondProject, 2023-11-20:
[rock-3a-radxa-5inch-touchscreen.dtbo.tar.gz](https://github.com/MichaIng/DietPi/files/13415549/rock-3a-radxa-5inch-touchscreen.dtbo.tar.gz)

Update: using this overlay on Debian OS (Raxda Official) the display works great and also the touch is acceptable. I've tried to copy and past this specific file attached on the "rockchip" folder and edit the "dietpiEnv" file adding this file in position "overlay=rock-3a-radxa-5inch-touchscreen.dtbo". Boot the Rock 3A but nothing change.
I hope that someone can help me to implement this overlay here.

Thanks

MichaIng, 2023-12-21:
This overlay is made for the legacy Linux 5.15 kernel provided by Radxa, not the mainline Linux 6.1 (or 6.6).

Setting HDMI display resolution for console mode is another question: #5641
Found some more possibilities here, but needs testing: https://github.com/MichaIng/DietPi/issues/6550#issuecomment-1677967216

MichaIng, 2025-02-09:
We added a new tool to control console modes and rotation via DRM/KMS sysfs and `video` kernel command-line parameter: #7390