MichaIng/DietPi #6741: Raspberry Pi 3B+ | System clock running progressively slower of NTP time

Trainax, 2023-11-10:
### Creating a bug report/issue

#### Required Information
- DietPi version

```
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=23
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='applied'
```

- Distro version

`bookworm 0`

- Kernel version

`Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`

- SBC model

`RPi 3 Model B+ (aarch64)`

- Power supply used
 
`5V 2.5A` (I don't remember the brand however I've never had undervoltage problems)

- SD card used

`SanDisk Ultra 64GB`

#### Steps to reproduce

1) Sync the system time manually using the `/boot/dietpi/func/run_ntpd 1` command (or wait for the automatic synchronization)
2) Observe the system time fall behind as time passes

#### Expected behaviour

The system time runs pretty much correctly between synchronizations using NTP (`Boot + Daily` mode selected in `dietpi-config`) 

#### Actual behaviour

The system clock falls pretty much rapidly out of sync with NTP time. I manually synced the system time with the `/boot/dietpi/func/run_ntpd 1` command at 18:04:47 CET and:
- at 18:07:27 the system time was 0.087492563 seconds slow of NTP time
- at 18:10:40 the system time was 0.197264552 seconds slow of NTP time
- at 18:24:39 the system time was 0.590976775 seconds slow of NTP time
- at 18:31:07 the system time was 0.797877610 seconds slow of NTP time
- at 18:39:45 the system time was 1.084610343 seconds slow of NTP time
- at 18:59:10 the system time was 1.694525599 seconds slow of NTP time
- at 19:03:27 the system time was 1.877578974 seconds slow of NTP time

In the previous days I've observed the system run up to 14 seconds slow of NTP time before the automatic daily synchronization

#### Extra details

- I used `debian.pool.ntp.org` as NTP server to sync the system time
- I run [cturra/docker-ntp](https://github.com/cturra/docker-ntp) to serve NTP requests inside my LAN using my Pi. This container runs without the privileged flag so it can't modify system time. The only thing which modifies the system clock is the daily NTP synchronization or when I manually run the system clock sync command like in this case. I've obtained the differences between the system times and NTP times by manually running the `docker exec ntp chronyc tracking` command over SSH at the times listed above
- I've never modified the CPU governor, done overclocking, ecc

Thank you in advance

Joulinar, 2023-11-10:
did you change voltage settings below 600?

Trainax, 2023-11-10:
> did you change voltage settings below 600?

No. I didn't modify anything of that kind

Joulinar, 2023-11-10:
usually this is a behavior if voltage has been set to low

MichaIng, 2023-11-10:
To be sure, can you paste the output of the `cpu` command? If min CPU frequency is indeed 600 MHz (less should actually not be allowed by the firmware), can you try to raise thr minimum frequency via `dietpi-config` performance options?

Another way would be to switch to `performance` CPU governor.

Trainax, 2023-11-10:
> To be sure, can you paste the output of the cpu command?

[image]

> can you try to raise the minimum frequency via dietpi-config performance options?

Yes, I can. What frequency should I set it to?

> Another way would be to switch to performance CPU governor

I can also try that

MichaIng, 2023-11-10:
Try 900 Mhz min for the start and see whether this helps already.

Trainax, 2023-11-12:
I've been running my Pi for the past day with the min frequency you suggested and it seems to have fixed the problem. The system clock is now consistently within ~0.3 seconds slow/fast of NTP time throughout the day.

I don't know if it has anything to do with it but it just occurred to me that the previous reboot (which was giving me this issue) happened because a breaker tripped and all my network devices lost power, so the Pi booted without an Internet connection (and the clock sync at starting time could not be completed) as both my router and my ONT were starting (the wired network interface was up because the Pi is connected to an Ethernet switch which starts pretty much instantaneously), while this boot (which is still up from when I rebooted after changing the performance options in `dietpi-config` a day ago as you suggested) happened with an Internet connection so the initial time synchronization completed successfully. It shouldn't make a difference as I manually synced the system time using the `/boot/dietpi/func/run_ntpd 1` command when I was having this issue, but I thought it might be worth reporting this detail

Joulinar, 2023-11-12:
Usually this will not have such impact. It's more related to CPU load and frequency settings.

MichaIng, 2023-11-12:
On RPi 4, it is possible to lower the minimal CPU frequency to 100 MHz, but this causes system clock to run very slow on most boards. 300 MHz was found to run fine on most boards, but there are still individual cases where the system clock runs too slow with this. 600 MHz is the default minimum, and the absolute minimum for all other RPi models, since more critical issues have been found for those: https://github.com/raspberrypi/firmware/issues/1431

It is hence very strange that this default causes issues in your case, but at least in principle the issue is known, which is why we thought about it in the first place when you reported your symptoms.

Is this a new RPi 3B+ or did you use this in the past without this issue? If it is new, and the same happens with the official RPi OS (only other possibly relevant issue is the CPU governor, still being `ondemand` on RPi OS, AFAIK), I would try to return the board, as it then seems to be an individual very bad quality chip.

Trainax, 2023-11-12:
>Is this a new RPi 3B+ or did you use this in the past without this issue?

No, I bought this Raspberry Pi in 2019 and it has been running 24/7 since mid-2020 (excluding some brief interruptions due to the power going out) because I use it (among other things) to host a DNS server for my LAN using AdGuard Home.

I only noticed this problem last Wednesday because I manually ran the `docker exec ntp chronyc tracking` command after installing [cturra/docker-ntp](https://github.com/cturra/docker-ntp) to use my Pi as an NTP server for the devices inside my LAN, so I don't know how long this problem has existed as I had never consistently checked the system time before. What I can tell you is that the system time never went wildly out of sync of NTP time otherwise DNSSEC queries wouldn't have been validated and it has never happened.

I've run my Pi with Raspberry Pi OS Lite from 2020 up to about 5 months ago when I switched to DietPi because a friend recommended it to me and I wanted to update my Pi to Bookworm. Unfortunately I can't tell if this problem also existed back then

Joulinar, 2023-11-12:
At last we found a workaround for time being.

Trainax, 2023-12-28:
I think this issue can be closed. The clock of Raspberry Pi 3B+ has been consistently running within 0.5 seconds of NTP time even using the `schedutil` governor for the past month. I didn't change anything but the governor to put it back to `schedutil` and the problem has not come back.

Thanks again!

closed 2023-12-28 by Trainax

MichaIng, 2023-12-29:
This is with 900 MHz minimum CPU frequency, right? Good that there is a workaround at least. I'll keep it in mind. In case there is another such case with any RPi 1/2/3 requiring a higher minimum frequency than default, we should report to RPi Ltd. Probably something can be done firmware-wise to at least assure that the default minimum works without slowing down the system clock.

Trainax, 2023-12-30:
>This is with 900 MHz minimum CPU frequency, right?

No, the minimum frequency can also be set back to 600 MHz without any problems (at least in my case)

[image]

System time was syncronized this morning at 01:25 as stated in the logs

[image]

and now, almost 9 hours later is still within 0.25 seconds of NTP time

[image]

MichaIng, 2023-12-30:
Ah okay that is great. No idea what happened for it to run slow when you opened the issue, also there was no kernel update in the meantime, but probably it was some unrelated strange thing, solved by another package upgrade or what.