MichaIng/DietPi #6742: `dietpi-installer` on LePotato pure-Debian Bookworm removes ethernet drivers

kx1t, 2023-11-10:
### Creating a bug report/issue

#### Required Information
- DietPi version | 8.23 RC 3
- Distro version | bookworm
- Kernel version | Linux DietPi 6.1.61-10985-g88acb5e7b94d #1 SMP PREEMPT_DYNAMIC Mon Nov  6 21:24:36 EST 2023 aarch64 GNU/Linux
- SBC model | Generic Device (aarch64) (LePotato)
- Power supply used | 5 Amps Anker
- SD card used | SanDisk ultra 64 Gb

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce
1. Install this firmware onto an SD: [debian-12-base-arm64+aml-s905x-cc.img.xz](https://distro.libre.computer/ci/debian/12/debian-12-base-arm64%2Baml-s905x-cc.img.xz)
2. Install OpenSSH-server and enable root login
3. Check that remote login works
4. Install DietPi using https://dietpi.com/docs/hardware/#make-your-own-distribution -  select either AmLogic S905 or Generic image (same effect with both)
5. Reboot

#### Expected behaviour
The ethernet device should be recognized as it was before

#### Actual behaviour
Ethernet device is not found

#### Extra details
Device simply not available in `/dev/` or with `ip a`. `ifup` doesn't make a difference, etc.

ManuVice, 2023-12-09:
same problem with Debian Bookworm in a lxc container. it removes ethernet driver..

MichaIng, 2023-12-20:
Yes, this can be an issue if another network stack is used on the pre-image. If it is NetworkManager, AFAIK it is sufficient to just re-add a DNS nameserver when facing this issue. When you run into an error can you try to "Open subshell" and there:
```sh
echo 'nameserver 9.9.9.9' > /etc/resolv.conf
exit
```
Then "Retry".

MichaIng, 2025-05-31:
I'll mark this as closed. Indeed I still need to find time to implement proper handling of NetworkManager and connman uninstals into our installer, which both tend to deconfigure the current network connection for whatever reason, sometimes DNS only. Tracked here: #2054

closed 2025-05-31 by MichaIng