MichaIng/DietPi #6743: cant install specific software combinations using dietpi-software

Ankharna, 2023-11-11:
### Creating a bug report/issue

#### Required Information
- DietPi version:
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=23
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='applied'

- Distro version:
bullseye 1

- Kernel version:
Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

- SBC model:
RPi 4 Model B (aarch64)

- Power supply used
Original - Raspberry Pi 4 Model B

- SD card used
120gb external ssd (no SD card installed)

#### Additional Information (if applicable)
- Software title
lighttpd, php, sqlite, nginx, apache, mariadb

- Was the software title installed freshly or updated/migrated?
LLSP [81] incl. sqlite, lighttpd, php was installed before
Nginx or Apache were not installed

- Can this issue be replicated on a fresh installation of DietPi?
likely

- Bug report ID:
308efec6-2857-41b1-aae0-881ce2be6330

#### Steps to reproduce
1. have LLSP [81] installed (dietpi-software)
2. try to install Nginx or Apache (dietpi-software)
3. in case of Nginx this warning pops up after checking the box and pressing enter to "lock in" that i intend to install Nginx:
```
[WARNING] Conflicting installs have been detected!                                           
                                                       │                                                                                           
                                                       │ Nginx won't be installed, as it conflicts with:                
                                                       │                                                                     
                                                       │  - LLSP: It is installed already.         
                                                       │                                                        
                                                       │  - LLMP: It is installed already.                                                                          
                                                       │                                                                                                                      
                                                       │  - Lighttpd: It is installed already.  
```
4. the checkmark is not applied -> installing nginx not possible

#### Expected behaviour
dietpi-software lets me install whatever i want in my desired configuration (want to try zabbix, all this software is required)

#### Actual behaviour
- error message pops up and doesnt explain why exactly these stacks are not compatible with (at least superficially) indepent software
- unchecks the intended software installation without obvious reason
- 
#### Extra details
- i tried a few variations to get the following sotware installed, using stacks and singular software installer (all within dietpi-software):
nginex, apache, mariadb, lighttpd, php, sqlite 
- its confusing that some software can only be found via software search, and is not listed in "browse software" (apache, php, lighttpd)
- im a noob in all this stuff, there might be connections i know nothing about, but maybe the error message could give a hint why something doesnt work in this case
- i got all necessary software installed and only need nginx or apache, but these two are problematic

(love dietpi, havnt had any issues until now. i opened this issue because chatgpt said all this software could be installed at the same time on a machine, but if theres some obvious connection im too blind to see im sorry to have wasted your time!)

MichaIng, 2023-11-11:
Lighttpd is a webserver like Nginx, both running on the same port 80 (and in case 443), hence naturally conflict. You would need to uninstall Lighttpd to install Nginx. But you likely also need to reinstall all web applications, to have their respective Nginx configs added.

Ankharna, 2023-11-11:
hey thank you for your fast response, and sorry for stirring up all this for me being unknowledgable.

i think my pihole is running on lighttpd, so i will have to figure this out, thank you for this hint!

closed 2023-11-11 by Ankharna

MichaIng, 2023-11-11:
Just for the record, also Pi-hole can run on Nginx, if you prefer it:
1. Uninstall Lighttpd:
```sh
dietpi-software uninstall 84
```
2. Reinstall Pi-hole: It will ask you for the webserver, where you can select Nginx:
```sh
dietpi-software reinstall 93
```

Ankharna, 2023-11-12:
thank you!

incredibly helpful advice! had to fix an issue with the time ive had for a month.
now i can delay containerizing pihole+unbound a bit further :D

your distro is the stuff! thank you!

cheers:)