MichaIng/DietPi #6745: REST Server

supertevran, 2023-11-12:
Restic is a fast and secure backup program. It's already included in DietPi's Software Repository.

Nevertheless it would be great to see a Rest Server Setup as a high performance HTTP server that implements restic's [REST backend API] https://restic.readthedocs.io/en/latest/100_references.html#rest-backend).

It provides secure and efficient way to backup data remotely, using [restic](https://github.com/restic/restic) backup client via the [rest: URL](https://restic.readthedocs.io/en/latest/030_preparing_a_new_repo.html#rest-server).

theAkito, 2023-11-13:
Why not [Borg](https://borgbackup.readthedocs.io/en/stable/), though?

Joulinar, 2023-11-13:
Maybe you like to connect and vote https://github.com/MichaIng/DietPi/discussions/5506

closed 2023-11-13 by Joulinar