MichaIng/DietPi #6747: Package raspi-firmware is not configured yet

piyushaswani55, 2023-11-13:
#### Details:
- Date           | Mon Nov 13 20:22:26 IST 2023
- DietPi version | v8.23.3 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux nebulae 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bookworm (ID=7,RASPBIAN=0)
- Command        | `apt-get -y --allow-change-held-packages install rpi-eeprom`
- Exit code      | 100
- Software title | DietPi-Set_hardware
#### Steps to reproduce:

1. Fresh install `DietPi_RPi-ARMv8-Bookworm` on RPi 4
2. This happens on first time setup
#### Expected behaviour:
The command should not error out
- ...
#### Actual behaviour:
rpi-eeprom is already the newest version (20.1-2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
2 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up raspi-firmware (1:1.20231024+ds-1+rpt1) ...
Error: missing /boot/firmware, did you forget to mount it?
dpkg: error processing package raspi-firmware (--configure):
 installed raspi-firmware package post-installation script subprocess returned error exit status 1
dpkg: dependency problems prevent configuration of rpi-eeprom:
 rpi-eeprom depends on raspi-firmware; however:
  Package raspi-firmware is not configured yet.

dpkg: error processing package rpi-eeprom (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 raspi-firmware
 rpi-eeprom
E: Sub-process /usr/bin/dpkg returned an error code (1)
- ...
#### Extra details:
Installed latest DietPi Bookworm on my RPi4.
- ...
#### Additional logs:
```

MichaIng, 2023-11-13:
That is bad, the latest version of `rpi-eeprom` depends on the new firmware package now, which requires a different boot mount setup. And the repository does not offer the old version anymore 😞.

Please select from the error-handle "Change command" and replace everything with `true`. Then after setup has finished, do this to install an older version manually and set the package on hold for now, until we figured out how to safely migrate RPi systems from old to new firmware packges:
```sh
cd /tmp
curl -fo package.deb 'https://archive.raspberrypi.org/debian/pool/main/r/rpi-eeprom/rpi-eeprom_20.1-1_all.deb'
apt install ./package.deb
rm package.deb
apt-mark hold rpi-eeprom
```

hrfried, 2023-11-14:
@MichaIng Can you confirm what you mean by 

> Please select from the error-handle "Change command" and replace everything with true.

Ran an `apt upgrade` and got a similar error and came here. And now I'm afraid to reboot as this pi is running my DHCP server, DNS server, pihole, rustdesk server, etc. I'm assuming if I purge and reinstall via above method, things should be okay but I'm confused what you meant by the above. Not something I have time to toy around with much right now lol. Seems like a pretty idiotic breaking change from upstream....sigh...

You've probably already seen, but if it's helpful for live-patch:
https://forums.raspberrypi.com/viewtopic.php?p=2156899
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1055901

matzepunkrocker, 2023-11-14:
Hello,

I ran into this right during post-configuration after a fresh install on my Pi4. There was indeed no /boot/firmware. If you look into the raspi-firmware package, the postinstall script says:

```
elif ! mountpoint -q /boot/firmware; then
      echo "Error: missing /boot/firmware, did you forget to mount it?" >&2
      exit 1
```

So I gave this script a mountpoint...

```
mkdir /boot/firmware
mount -t tmpfs -o size=128M firmware /boot/firmware
```

...and it happily installed :-)

While this is only a tmpfs, I made it a real folder afterwards

```
cp -r /boot/firmware /tmp/
umount /boot/firmware
rm -r /boot/firmware
cp -r /tmp/firmware /boot/firmware
```

Will someone please enlighten me? Why should /boot/firmware be a mountpoint at all?

MichaIng, 2023-11-14:
See here: https://github.com/MichaIng/DietPi/issues/6676#issuecomment-1771124218

Unintentionally the new `rpi-eeprom` package pulls this new `raspi-firmware` package as dependency, which cannot work without the partitioning/mount change. The RPi package maintainers forgot about those which manually upgraded to Bookworm, hence use the old `/boot` mount, or 3rd party distros like us.

An unclean migrating can cause havoc, hence as long as you are not sure about what you are doing, I strongly advice to apply the workaround I posted above, which downgrades and holds the `rpi-eeprom` package, and purges the new `raspi-firmware` package for now.

The migration according to the linked RPi forum topic is possible, but incomplete: You need to purge the old bootloader package and migrate the kernel package(s) as well to assure that everything is compatible with each other. Else any kernel and/or bootloader upgrade or reinstall might break things.

I am writing a script for a clean and complete migration, which will be offered on next DietPi update.

MichaIng, 2023-11-16:
Okay, I merged a live patch + pre-patch to prevent this package from being upgraded + install the latest compatible version when done via `dietpi-config`. RPi images are also rebuilding with the patch applied already.

The actual migration to the new firmware/kernel packages can be discussed here: #6676
They are required for RPi 5 support as well.

closed 2023-11-16 by MichaIng

whyisthisbroken, 2023-11-16:
when will the live-patch be online ? Also in 8.24 or only 8.23 ?

MichaIng, 2023-11-16:
It is online already. Just run `dietpi-update` and it will be offered. All it does is setting the package on hold and adjusting the `dietpi-config` (and first boot) option so that it installs the last compatible version of the package hosted on our server, instead of the new one. In 8.24 (`dev` branch) it has been solved code-wise the regular way.

Btw, I reported it to the RPi repo as well, and they removed the dependency, with a note that the package should go into the RPi repo next week. Afterwards, the workaround is not required anymore 🙂.

whyisthisbroken, 2023-11-16:
Okay, thanks for info. The live-patch doesnt show up, i dont know why.

> root@HomeNetDNS:~# dietpi-update
> 
>  DietPi-Update
> ─────────────────────────────────────────────────────
>  Phase: Checking for available DietPi update
> 
> [  OK  ] DietPi-Update | Checking IPv4 network connectivity
> [  OK  ] DietPi-Update | Checking IPv6 network connectivity
> [  OK  ] DietPi-Update | Checking DNS resolver
> [ INFO ] DietPi-TimeSync | Manual mode detected, skipping time sync
> [ INFO ] DietPi-Update | Getting latest version from: https://raw.githubusercontent.com/MichaIng/DietPi/beta/.update/version
> [  OK  ] DietPi-Update | Got valid latest version: 8.24.0
> [  OK  ] DietPi-Update | No update required, your DietPi installation is already up to date:
> [ INFO ] DietPi-Update | Current version : v8.24.0
> [ INFO ] DietPi-Update | Latest version  : v8.24.0
> [ INFO ] DietPi-Update | Checking for new available live patches
> [ INFO ] DietPi-Update | APT update, please wait...
> Hit:1 https://deb.debian.org/debian bookworm InRelease
> Hit:2 https://deb.debian.org/debian bookworm-updates InRelease
> Hit:3 https://deb.debian.org/debian-security bookworm-security InRelease
> Hit:4 https://deb.debian.org/debian bookworm-backports InRelease
> Hit:5 https://archive.raspberrypi.org/debian bookworm InRelease
> Hit:6 https://packages.redis.io/deb bookworm InRelease
> Reading package lists...
> [  OK  ] DietPi-Update | APT update
> [ INFO ] DietPi-Update | No APT upgrades were found, not creating file: /run/dietpi/.apt_updates
> 

If i run "apt upgrade" i get the message "2 not fully installed or removed." that are raspi-eeprom and raspi-firmware...
"apt update" says it all up2date... strange behavior... i dont wanna reset my entire setup :(((

MichaIng, 2023-11-16:
> If i run "apt upgrade" i get the message "2 not fully installed or removed." that are raspi-eeprom and raspi-firmware...
"apt update" says it all up2date... strange behavior... i dont wanna reset my entire setup :(((

Please apply the workaround I provided above: https://github.com/MichaIng/DietPi/issues/6747#issuecomment-1808348281
The live patch only works if the package upgrade has not been attempted yet.

> Okay, thanks for info. The live-patch doesnt show up, i dont know why.

Because you are on v8.24 already (beta or dev?). If you are on beta, wait for v8.24.1, if you are on dev:
```sh
dietpi-update -1
```

whyisthisbroken, 2023-11-16:
Okay thanks, i wait for 8.24.1 then and hope that the livepatch works, even if I have already initiated the package upgrade.
Finger crossed :)

MichaIng, 2023-11-16:
> Please apply the workaround I provided above: https://github.com/MichaIng/DietPi/issues/6747#issuecomment-1808348281
The live patch only works if the package upgrade has not been attempted yet.

BerndKohl, 2023-11-16:
Sorry, but how do I check if all is good now?
I ran apt update/upgrade BEFORE noticing that there was a LIVE patch.
What I did:
1. ran apt update/upgrade /reports known error
2. applied LIVE patch
3. applied workaround mentioned above...

What can I do to verfiy that my RPi will still boot up after a shutdown?

best

Bernd

Joulinar, 2023-11-16:
The system will boot still. There should be nothing to worry about.

Johnex, 2023-11-19:
#

> That is bad, the latest version of `rpi-eeprom` depends on the new firmware package now, which requires a different boot mount setup. And the repository does not offer the old version anymore 😞.
> 
> Please select from the error-handle "Change command" and replace everything with `true`. Then after setup has finished, do this to install an older version manually and set the package on hold for now, until we figured out how to safely migrate RPi systems from old to new firmware packges:
> 
> ```shell
> G_AGP rpi-eeprom raspi-firmware
> cd /tmp
> curl -fo package.deb 'https://archive.raspberrypi.org/debian/pool/main/r/rpi-eeprom/rpi-eeprom_20.1-1_all.deb'
> apt install ./package.deb
> rm package.deb
> apt-mark hold rpi-eeprom
> ```

Thanks for this fix!
I also had to remove the raspi-firmware package, or else the latest dietpi upgrade did not complete, it would fail on apt because the firmware package was not configured.

MichaIng, 2023-11-20:
A new version of `rpi-eeprom` has been pushed to APT servers, which does not depend on `raspi-firmware` anymore. Hence it is again possible to unhold and upgrade it:
```sh
apt-mark unhold rpi-eeprom
apt update
apt upgrade
```
Will be done next DietPi update as well.

evrial, 2023-12-03:
So who is to blame here, his name?

evrial, 2023-12-03:
I want to make funny jokes on him when this happens again.

Joulinar, 2023-12-03:
What a bad attitude to make fun of open source projects where people sacrifice their free time for people like you who then have the products for free. We are all human and mistakes can happen. It would be better if you actively participated in the relevant projects yourself instead of making fun of them. You should be ashamed of yourself.

evrial, 2023-12-03:
That's an interesting card you pulled out. I should be thankful to profitable entity who sells 4 year old credit card sized computers for 100-200% markup. OK

whyisthisbroken, 2023-12-03:
i miss the feature to mute such idiots on github...

mikeinanaheim, 2023-12-04:
> I want to make funny jokes on him when this happens again.

Why TF would you think it is fun to go after people who create cool tools and WORK FOR YOU FOR FREE??

srs4511351, 2023-12-25:
I am running Raspberry Pi 4B

I installed the image released on 2023-11-23 and upgraded it to DietPi v8.25.1 and ran the firmware migration script.
Everything works, but there seems to be a problem with the vc4-fkms-v3d driver.
SDR++ has a high CPU usage when it is idle (not playing an SDR). I also reports a very low frame rate.
Changing to boot from another system upgraded to DietPi v8.25.1, CPU usage is low.

This had been identified as a driver issue, and the OpenGL version that is used. The vc4-fkms-v3d driver dramatically reduced CPU usage.
Please see SDR++ [Graphics issue with Raspberry Pi 4 #844](https://github.com/AlexandreRouma/SDRPlusPlus/issues/844) 

I should note that newer systems work well with either the vc4-kms-v3d or the vc4-fkms-v3d driver.
Raspberry Pi Bookworm OS no longer allows the vc4-fkms-v3d driver.

Are the vc4-kms-v3d drivers being loaded correctly?

dirkhh, 2024-01-02:
It seems that an update from v8.23.3 to v8.25.1 can still trigger this bug - and I don't understand how that's possible.
I had a user of a 8.23.3 based image fall into that trap today and it took a while to walk them remotely through getting this resolved.

Is there a step that I can do prior to running `dietpi-update 1` that makes sure this doesn't happen?

draxel, 2024-01-02:
> It seems that an update from v8.23.3 to v8.25.1 can still trigger this bug - and I don't understand how that's possible. I had a user of a 8.23.3 based image fall into that trap today and it took a [...]

I'm unfortunately another victim of the 8.23.3 -> 8.25.1 trap >.> As of today what is the recommended guidance for getting the upgrade to finish up properly if I've encountered this bug?

Thanks in advance!

Joulinar, 2024-01-02:
Can you share the whole complete error message you have on screen?

draxel, 2024-01-02:
Happily - this is (I believe) the relevant bit that I get in the terminal (let me know if there's anything else you're looking for):

```
[ INFO ] DietPi-Update | APT upgrade, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
                                                              1 not fully installed or removed.
               After this operation, 0 B of additional disk space will be used.
                                                                               Setting up raspi-firmware (1:1.20231024+ds-1+rpt2) ...
stat: cannot read file system information for '/boot/firmware': No such file or directory
Error: missing /boot/firmware, did you forget to mount it?
dpkg: error processing package raspi-firmware (--configure):
 installed raspi-firmware package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 raspi-firmware
E: Sub-process /usr/bin/dpkg returned an error code (1)
[FAILED] DietPi-Update | APT upgrade
 - Command: apt-get -y --with-new-pkgs upgrade
```

MichaIng, 2024-01-02:
```
G_AGP raspi-firmware
```

draxel, 2024-01-02:
Yep, that unclogged it!  Appreciate the assistance o7