MichaIng/DietPi #6748: I can't install Radarr anymore.

sagecandida, 2023-11-13:
#### Details:
- Date           | Mon Nov 13 20:10:34 CET 2023
- DietPi version | v8.23.3 (MichaIng/master)
- Image creator  | 
- Pre-image      | 
- Hardware       | RPi Zero 2 W (armv7l) (ID=3)
- Kernel version | `Linux DietPi 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux`
- Distro         | bookworm (ID=7,RASPBIAN=1)
- Command        | `tar xf Radarr.master.5.1.3.8246.linux-core-arm.tar.gz`
- Exit code      | 2
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```

```

MichaIng, 2023-11-13:
I just tried it here, and it works well.

Probably the download was corrupt. Can you retry it? If it fails again at the same point, are there some logs given on the error handler, or after exiting the script?

sagecandida, 2023-11-13:
[image]

[image]

I tried again and still doesn't work.

sagecandida, 2023-11-13:
I tried updating to beta version to see if it works on beta but not a chance.

MichaIng, 2023-11-13:
Hmm, strange, no logs at all. Can you try this manually:
```sh
cd /tmp
curl -fLO 'https://github.com/Radarr/Radarr/releases/download/v5.1.3.8246/Radarr.master.5.1.3.8246.linux-core-arm.tar.gz'
tar xf Radarr.master.5.1.3.8246.linux-core-arm.tar.gz
ls -l Radarr
```

sagecandida, 2023-11-13:
root@DietPi:/tmp# tar xf Radarr.master.5.1.3.8246.linux-core-arm.tar.gz
tar: Radarr/libcoreclr.so: Wrote only 5120 of 10240 bytes
tar: Radarr/Instances.dll: Cannot write: No space left on device
tar: Radarr/System.Reflection.dll: Cannot write: No space left on device
tar: Radarr/Microsoft.Extensions.FileProviders.Physical.dll: Cannot write: No space left on device
tar: Radarr/ICSharpCode.SharpZipLib.dll: Cannot write: No space left on device
tar: Radarr/System.IO.Pipes.AccessControl.dll: Cannot write: No space left on device
tar: Radarr/System.Text.Encoding.Extensions.dll: Cannot write: No space left on device
tar: Radarr/Microsoft.AspNetCore.Authorization.Policy.dll: Cannot write: No space left on device
tar: Radarr/Microsoft.AspNetCore.Mvc.Core.dll: Cannot write: No space left on device
tar: Radarr/Microsoft.Extensions.Hosting.WindowsServices.dll: Cannot write: No space left on device
tar: Radarr/System.Security.Principal.dll: Cannot write: No space left on device
tar: Radarr/System.IO.Compression.Brotli.dll: Cannot write: No space left on device
tar: Radarr/System.Private.CoreLib.dll: Cannot write: No space left on device
tar: Radarr/FluentMigrator.Runner.SqlServer.dll: Cannot write: No space left on device
tar: Radarr/BouncyCastle.Crypto.dll: Cannot write: No space left on device
tar: Radarr/libSystem.Native.so: Cannot write: No space left on device
tar: Radarr/Microsoft.Extensions.Logging.Debug.dll: Cannot write: No space left on device
tar: Radarr/System.Private.Xml.Linq.dll: Cannot write: No space left on device
tar: Radarr/Microsoft.IdentityModel.Logging.dll: Cannot write: No space left on device
tar: Radarr/Microsoft.Extensions.Caching.Abstractions.dll: Cannot write: No space left on device
tar: Radarr/Microsoft.AspNetCore.Localization.Routing.dll: Cannot write: No space left on device
tar: Radarr/System.Runtime.Serialization.dll: Cannot write: No space left on device
tar: Radarr/Microsoft.AspNetCore.Http.dll: Cannot write: No space left on device
tar: Radarr/System.Data.dll: Cannot write: No space left on device
tar: Radarr/System.Runtime.InteropServices.RuntimeInformation.dll: Cannot write: No space left on device
tar: Radarr/Microsoft.AspNetCore.Http.Results.dll: Cannot write: No space left on device
tar: Radarr/Microsoft.VisualBasic.dll: Cannot write: No space left on device
tar: Radarr/Microsoft.Extensions.Configuration.Binder.dll: Cannot write: No space left on device
tar: Radarr/System.Drawing.Primitives.dll: Cannot write: No space left on device
tar: Radarr/Diacritical.dll: Cannot write: No space left on device
tar: Radarr/Microsoft.AspNetCore.SignalR.Core.dll: Cannot write: No space left on device
tar: Radarr/System.Net.WebSockets.Client.dll: Cannot write: No space left on device
tar: Radarr/Microsoft.Extensions.Identity.Stores.dll: Cannot write: No space left on device
tar: Radarr/System.IO.Compression.FileSystem.dll: Cannot write: No space left on device
tar: Radarr/System.Diagnostics.FileVersionInfo.dll: Cannot write: No space left on device
tar: Radarr/Microsoft.Extensions.Diagnostics.HealthChecks.Abstractions.dll: Cannot write: No space left on device
tar: Radarr/Microsoft.AspNetCore.Hosting.dll: Cannot write: No space left on device
tar: Radarr/Microsoft.Extensions.Primitives.dll: Cannot write: No space left on device
tar: Radarr/Microsoft.Extensions.Hosting.Abstractions.dll: Cannot write: No space left on device
tar: Radarr/Microsoft.Data.SqlClient.dll: Cannot write: No space left on device
tar: Radarr/Microsoft.Win32.Primitives.dll: Cannot write: No space left on device
tar: Radarr/System.Security.Cryptography.Xml.dll: Cannot write: No space left on device
tar: Radarr/System.ComponentModel.DataAnnotations.dll: Cannot write: No space left on device
tar: Radarr/Microsoft.AspNetCore.Components.Web.dll: Cannot write: No space left on device
tar: Radarr/DryIoc.dll: Cannot write: No space left on device
tar: Radarr/System.Xml.XDocument.dll: Cannot write: No space left on device
tar: Radarr/Microsoft.AspNetCore.WebSockets.dll: Cannot write: No space left on device
tar: Radarr/Microsoft.AspNetCore.HttpOverrides.dll: Cannot write: No space left on device
tar: Radarr/System.Net.WebProxy.dll: Cannot write: No space left on device
tar: Radarr/UI/420-de64e22271b9062a68e8.js: Cannot write: No space left on device
tar: Radarr/UI/login.html: Cannot write: No space left on device
tar: Radarr/UI/189-384f51e46762242ce177.js.map: Cannot write: No space left on device
tar: Radarr/UI/420-de64e22271b9062a68e8.js.LICENSE.txt: Cannot write: No space left on device
tar: Radarr/UI/420-de64e22271b9062a68e8.js.map: Cannot write: No space left on device
tar: Radarr/UI/index-f281bf95d8dab4159671.js.map: Cannot write: No space left on device
tar: Radarr/UI/oauth.html: Cannot write: No space left on device
tar: Radarr/UI/index.html: Cannot write: No space left on device
tar: Radarr/UI/index-f281bf95d8dab4159671.js: Cannot write: No space left on device
tar: Radarr/UI/Content/189-59ceffe4140b5e6e5bf7.css: Cannot write: No space left on device
tar: Radarr/UI/Content/189-59ceffe4140b5e6e5bf7.css.map: Cannot write: No space left on device
tar: Radarr/UI/Content/Images/logo.svg: Cannot write: No space left on device
tar: Radarr/UI/Content/Images/logo-full.png: Cannot write: No space left on device
tar: Radarr/UI/Content/Images/Icons/logo-radarr.png: Cannot write: No space left on device
tar: Radarr/UI/Content/Images/Icons/logo-lidarr.png: Cannot write: No space left on device
[... cut ...]
tar: Radarr/Radarr.Host.deps.json: Cannot write: No space left on device
tar: Radarr/System.Threading.Tasks.Extensions.dll: Cannot write: No space left on device
tar: Radarr/Mono.Posix.NETStandard.dll: Cannot write: No space left on device
tar: Radarr/Microsoft.Extensions.FileProviders.Embedded.dll: Cannot write: No space left on device
tar: Radarr/Microsoft.AspNetCore.Server.Kestrel.Transport.Sockets.dll: Cannot write: No space left on device
tar: Exiting with failure status due to previous errors

sagecandida, 2023-11-13:
It must be something wrong because here is my disk usage
[image]

MichaIng, 2023-11-13:
I had such a feeling that insufficient memory is the reason on a 512 MiB RAM RPi. It is downloaded to and extracted within `/tmp`, which is a tmpfs (RAM disk). Please add a swap file:
```sh
/boot/dietpi/func/dietpi-set_swapfile 512
```
Theoretically you can remove it again afterwards
```sh
/boot/dietpi/func/dietpi-set_swapfile 0
```
But it does not hurt to leave it in place. Thanks to `swappiness=1` it is really only used if really needed. And Radarr alone already requires 200 MiB or so in idle. Now you want a download server on top, and both being active concurrently, that would not work with 475 MiB memory anyway.

sagecandida, 2023-11-13:
It works now. Thank you! And yes, you are right. It's hard to make it work due to lack of memory but i still wanted to give it a try.

closed 2023-11-13 by sagecandida

MichaIng, 2023-11-13:
The swap file makes it possible, and I don't think that those additional memory is used too regularly to affect performance significantly. Probably good to keep an eye on `htop` or `free` by times to verify that. If it is more regularly used, keep in mind that a (heavily used) swap file on SD card is not only bad performance-wise, but also degrades it life time. Better to attach a USB stick/drive and put the swap file there, in this case.