MichaIng/DietPi #6751: Error: missing /boot/firmware, did you forget to mount it?

realAbitbol, 2023-11-14:
### Creating a bug report/issue

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=23
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='applied'`
- Distro version | `bookworm 0`
- Kernel version | `Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- SBC model | `RPi 4 Model B (aarch64)`
- Power supply used | `official raspberry pi 4 alimentation`
- SD card used | `Sandisk Ultra`

#### Additional Information (if applicable)
- Bug report ID | `4067a938-6d2c-4e5f-ad7e-9f373d8adbed`

#### Steps to reproduce
1. `apt update; apt upgrade`

#### Expected behaviour
Upgrade should be fine

#### Actual behaviour
`Paramétrage de raspi-firmware (1:1.20231024+ds-1+rpt1) ...
**Error: missing /boot/firmware, did you forget to mount it?**`

#### Extra details
- Manage to "fix" the problem by removing raspi-firmware and raspi-eeprom packages
```
dietpi@raspberrypi:~$ ls /boot
bcm2710-rpi-2-b.dtb	 bcm2711-rpi-4-b.dtb	 COPYING.linux       fixup4.dat    kernel8.img       start4db.elf
bcm2710-rpi-3-b.dtb	 bcm2711-rpi-cm4.dtb	 dietpi	     fixup4db.dat  LICENCE.broadcom  start4.elf
bcm2710-rpi-3-b-plus.dtb  bcm2711-rpi-cm4-io.dtb  dietpi-LICENSE.txt  fixup4x.dat   overlays	     start4x.elf
bcm2710-rpi-cm3.dtb	 bcm2711-rpi-cm4s.dtb	 dietpi-README.md    fixup_cd.dat  pieeprom.sig      start_cd.elf
bcm2710-rpi-zero-2.dtb	 bootcode.bin		 dietpi.txt	     fixup.dat     pieeprom.upd      start_db.elf
bcm2710-rpi-zero-2-w.dtb  cmdline.txt		 dietpi-wifi.txt     fixup_db.dat  RECOVERY.000      start.elf
bcm2711-rpi-400.dtb	 config.txt		 fixup4cd.dat	     fixup_x.dat   start4cd.elf      start_x.elf
```

realAbitbol, 2023-11-14:
Gonna try to refresh from scratch, will report.

MichaIng, 2023-11-14:
Check out this topic, including a temporary workaround: #6747

closed 2023-11-14 by MichaIng

realAbitbol, 2023-11-14:
Thanks for the workaround !