MichaIng/DietPi #6753: Wireguard fails on Odroid XU4

pranavt, 2023-11-15:
I have managed to get wireguard working earlier without any issue whatsoever, so this seems to be a regression since I've searched for and found issues with Odroids and wireguard in the Issues list. Installed Wireguard by simply doing `sudo apt install wireguard resolvconf` (after apt update and apt upgrade). All further information included below.

#### Required Information
- DietPi version

`G_DIETPI_VERSION_CORE=8`
`G_DIETPI_VERSION_SUB=23`
`G_DIETPI_VERSION_RC=3`
`G_GITBRANCH='master'`
`G_GITOWNER='MichaIng'`
`G_LIVE_PATCH_STATUS[0]='applied'`
`G_LIVE_PATCH_STATUS[1]='applied'`

- Distro version
`bookworm`

- Kernel version
`Linux DietPi 5.4.253-current-odroidxu4 #2 SMP PREEMPT Mon Aug 14 01:33:04 UTC 2023 armv7l GNU/Linux`

- SBC model
`Odroid XU3/XU4/MC1/HC1/HC2 (armv7l)` (It is an XU4)

- Power supply used
`5V 6A`

- SD card used
`SanDisk Ultra 32GB`

#### Additional Information (if applicable)
- Software title | Wireguard
- Was the software title installed freshly or updated/migrated? - Fresh Install
- Can this issue be replicated on a fresh installation of DietPi? - Yes, it was a fresh Dietpi install on my XU4

#### Steps to reproduce
1. Install Dietpi
2. `sudo apt install wireguard resolvconf`
3. Import a working wireguard config, place it in /etc/wireguard/wg0.conf
4. Run `sudo wg-quick up wg0`

#### Expected behaviour
- The wg0 interface should come up

#### Actual behaviour
`[#] ip link add wg0 type wireguard`
`Error: Unknown device type.`
`Unable to access interface: Protocol not supported`
`[#] ip link delete dev wg0`
`Cannot find device "wg0"`

#### Extra details

`dietpi@DietPi:~$ dpkg -l | grep 'linux-image'`
`ii  linux-image-6.1.0-13-rt-armmp  6.1.55-1 armhf        Linux 6.1 for ARMv7 multiplatform compatible SoCs, PREEMPT_RT`
`ii  linux-image-current-odroidxu4  23.8.1 armhf        Armbian Linux current kernel image 5.4.253-current-odroidxu4`
`ii  linux-image-rt-armmp           6.1.55-1 armhf        Linux for ARMv7 multiplatform compatible SoCs (meta-package)`

`dietpi@DietPi:~$ ls -al /lib/modules`
`total 16`
`drwxr-xr-x  4 root root 4096 Nov 16 00:57 .`
`drwxr-xr-x 46 root root 4096 Nov 16 00:58 ..`
`drwxr-xr-x  3 root root 4096 Oct 29 04:45 5.4.253-current-odroidxu4`
`drwxr-xr-x  3 root root 4096 Nov 16 00:58 6.1.0-13-rt-armmp`

`dietpi@DietPi:~$ ls -al /boot/`
`total 81332`
`drwxr-xr-x  4 root root     4096 Nov 16 00:59 .`
`drwxr-xr-x 18 root root     4096 Oct 29 04:55 ..`
`-rw-r--r--  1 root root  3132347 Aug 30 06:43 System.map-5.4.253-current-odroidxu4`
`-rw-r--r--  1 root root       83 Sep 29 09:45 System.map-6.1.0-13-rt-armmp`
`-rw-r--r--  1 root root    10803 Oct 29 04:40 boot.ini`
`-rw-r--r--  1 root root   168797 Aug 30 06:43 config-5.4.253-current-odroidxu4`
`-rw-r--r--  1 root root   273925 Sep 29 09:45 config-6.1.0-13-rt-armmp`
`drwxr-xr-x  4 root root     4096 Nov 15 23:08 dietpi`
`-rw-r--r--  1 root root    18092 Oct 25 02:29 dietpi-LICENSE.txt`
`-rw-r--r--  1 root root    16059 Oct 25 02:29 dietpi-README.md`
`-rw-------  1 root root     3950 Oct 29 04:55 dietpi-wifi.txt`
`-rw-r--r--  1 root root    17465 Nov 15 23:09 dietpi.txt`
`lrwxrwxrwx  1 root root       29 Oct 29 04:46 dtb -> dtb-5.4.253-current-odroidxu4`
`drwxr-xr-x  3 root root     4096 Oct 29 04:45 dtb-5.4.253-current-odroidxu4`
`-rw-r--r--  1 root root  8189087 Oct 29 04:46 initrd.img-5.4.253-current-odroidxu4`
`-rw-r--r--  1 root root 25180211 Nov 16 00:59 initrd.img-6.1.0-13-rt-armmp`
`lrwxrwxrwx  1 root root       25 Nov 16 00:59 uInitrd -> uInitrd-6.1.0-13-rt-armmp`
`-rw-r--r--  1 root root  8189151 Oct 29 04:46 uInitrd-5.4.253-current-odroidxu4`
`-rw-r--r--  1 root root 25180275 Nov 16 00:59 uInitrd-6.1.0-13-rt-armmp`
`-rw-r--r--  1 root root  7391656 Aug 30 06:43 vmlinuz-5.4.253-current-odroidxu4`
`-rw-r--r--  1 root root  5464576 Sep 29 09:45 vmlinuz-6.1.0-13-rt-armmp`
`lrwxrwxrwx  1 root root       33 Oct 29 04:46 zImage -> vmlinuz-5.4.253-current-odroidxu4`

Joulinar, 2023-11-16:
A little bit as expected because you are running kernel version below 5.6. Means you are missing the required the WireGuard kernel module. Only option on these legacy kernel version would be to use DKMS builds.

At least from our side we removed support for DKMS builds with DietPi 8.12 https://github.com/MichaIng/DietPi/commit/e8a944a. 

However you could try to build WireGuard kernel module yourself.

pranavt, 2023-11-16:
Thanks for the information, I had missed the change since I hadn't been running my XU4 for a while now.

I have managed to compile the kernel module and Wireguard is running now. If anyone stumbles upon this, here is what I did:

1. Removed the earlier wireguard install along with wireguard-tools
2. Installed kernel headers - `sudo apt install linux-headers-current-odroidxu4`
3. Installed the following packages: `sudo apt install libelf-dev build-essential pkg-config`
4. Rest of the instructions from the Wireguard compile page here: https://www.wireguard.com/compilation/ (Start at step 2)

After this, bringing up the wireguard interface works as normal. I guess this will not be fixed in the future, so there's no point leaving this issue open.

closed 2023-11-16 by pranavt

MichaIng, 2023-11-16:
Generally, also `wireguard-dkms` can be used, which automatically rebuilds the module on every kernel upgrade.

Strange that the module is missing. It was present in an earlier version of the XU4 kernel. Armbian backported the WireGuard module into all 5.4 kernel version that time, when it was latest LTS.

I am testing a build with the module enabled and in case open a PR at Armbian to re-add it.

MichaIng, 2023-11-16:
Oh lol, Armbian moved to Linux 6.1 with the board: https://github.com/armbian/build/commit/161d58c
The only question is how long it takes now until 23.11.y builds are uploaded to the APT repository. For 23.08.y builds it took more than half a year. Pressing thumbs.

MichaIng, 2023-11-18:
Here are kernel packages with the WireGuard module added: https://dietpi.com/downloads/binaries/testing/
Works pretty well, and is Linux 6.1 btw. Armbian plans a release on November 26th, so I hope that Linux 6.1 will be pushed for the XU4 to APT servers at this date as well. And I'll open a PR, which will be hopefully merged before that, to have WireGuard included directly.

MichaIng, 2023-11-19:
My PR was merged. So next kernel release should contain the WireGuard module 🙂.

pranavt, 2023-11-20:
Thank you so much!

MichaIng, 2023-12-22:
I just checked, and the module is now present when `apt upgrade`ing to the latest kernel package.