MichaIng/DietPi #6755: dietpi Firmware error

joshi0531, 2023-11-16:
#### Details:
- Date           | Thu Nov 16 20:34:17 CET 2023
- DietPi version | v8.23.3 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux my-pi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bookworm (ID=7,RASPBIAN=0)
- Command        | `apt-get -y --allow-change-held-packages install ./rpi-eeprom.deb`
- Exit code      | 100
- Software title | DietPi-Set_hardware
#### Steps to reproduce:
1. dietpi-config
2. Advanced Option
3. Update RPi4 EEPROM firmware
#### Expected behaviour:

- fix of the frimware issue for
#### Actual behaviour:

- Repair of the error message when installing new software.  
#### Extra details:

- ...
#### Additional logs:
```
Reading package lists...
```

Joulinar, 2023-11-16:
See https://github.com/MichaIng/DietPi/issues/6747#issuecomment-1808348281

MichaIng, 2023-11-16:
The live patch has been applied already (as it is installing a local `deb` package. I guess you ran into the error already before applying the live patch? It does not work if you did run into the error already. In this case, you need to purge the newly pulled firmware package manually once:
```sh
G_AGP raspi-firmware
```

joshi0531, 2023-11-16:
yes it works, but i wonder why this happened. i have three raspi's walking with diet pi and only my big nas pi is affected.

Joulinar, 2023-11-16:
It's affecting RPi4 running Debian Bookworm only. And it depends on which stage the live patch has been applied.