MichaIng/DietPi #6757: rdtclient

dario19720108, 2023-11-17:
### Creating a software request

#### Formal software information
- Software title | 
- Short description | 
- Official URL (if available) | 
- GitHub URL (if available) | 
- (Official) install documentation | 

#### Are there similar/alternative software titles available with DietPi-Software?

- ...

#### What makes your requested software better than the above solutions, if available?

- ...

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?

- ...

#### Can you provide the installation steps that you would suggest DietPi-Software to do?

1. ...

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.

- ...

dario19720108, 2023-11-17:
This is a web interface to manage your torrents on Real-Debrid, AllDebrid or Premiumize. It supports the following features:

Add new torrents through magnets or files
Download all files from Real-Debrid, AllDebrid or Premiumize to your local machine automatically
Unpack all files when finished downloading
Implements a fake qBittorrent API so you can hook up other applications like Sonarr, Radarr or Couchpotato.
Built with Angular 15 and .NET 8
You will need a Premium service at Real-Debrid, AllDebrid or Premiumize!

MichaIng, 2023-11-18:
Duplicate of #6759. Let's keep the other one with more information.

closed 2023-11-18 by MichaIng