MichaIng/DietPi #6763: Error upgrade

pogoarte, 2023-11-19:
DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for available DietPi update

[  OK  ] DietPi-Update | Checking IPv4 network connectivity
[  OK  ] DietPi-Update | Checking DNS resolver
[ INFO ] DietPi-Update | Getting latest version from: https://raw.githubusercontent.com/MichaIng/DietPi/master/.update/version
[  OK  ] DietPi-Update | Got valid latest version: 8.24.1
[  OK  ] DietPi-Update | Update available:
[ INFO ] DietPi-Update | Current version : v8.23.3
[ INFO ] DietPi-Update | Latest version  : v8.24.1

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for update pre-requirements

[  OK  ] DietPi-Update | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Update | Free space check: path=/ | available=27391 MiB | required=100 MiB
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Applying pre-patches

[  OK  ] DietPi-Update | Downloading pre-patches
[  OK  ] DietPi-Update | Applying execute permission
[ INFO ] DietPi-Pre-patches | Setting rpi-eeprom package on hold to prevent conflicting upgrade attempt
[  OK  ] DietPi-Pre-patches | apt-mark hold rpi-eeprom
[  OK  ] DietPi-Update | Successfully applied pre-patches

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Upgrading APT packages

[ INFO ] DietPi-Update | APT update, please wait...
Hit:1 https://deb.debian.org/debian bookworm InRelease
Hit:2 https://deb.debian.org/debian bookworm-updates InRelease
Hit:3 https://deb.debian.org/debian-security bookworm-security InRelease
Hit:4 https://deb.debian.org/debian bookworm-backports InRelease
Hit:5 https://archive.raspberrypi.org/debian bookworm InRelease
Reading package lists...
[  OK  ] DietPi-Update | APT update
[ INFO ] DietPi-Update | APT upgrade, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
                                                              2 not fully installed or removed.
                                                                                               After this operation, 0 B of additional disk space will be used.
                                                                                                                                                               Setting up raspi-firmware (1:1.20231024+ds-1+rpt2) ...
stat: cannot read file system information for '/boot/firmware': No such file or directory
Error: missing /boot/firmware, did you forget to mount it?
dpkg: error processing package raspi-firmware (--configure):
 installed raspi-firmware package post-installation script subprocess returned error exit status 1
dpkg: dependency problems prevent configuration of rpi-eeprom:
 rpi-eeprom depends on raspi-firmware; however:
  Package raspi-firmware is not configured yet.

dpkg: error processing package rpi-eeprom (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 raspi-firmware
 rpi-eeprom
E: Sub-process /usr/bin/dpkg returned an error code (1)
[FAILED] DietPi-Update | APT upgrade
 - Command: apt-get -y --with-new-pkgs upgrade

#### Details:
- Date           | Sun Nov 19 09:46:20 CET 2023
- DietPi version | v8.23.3 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux mecap 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bookworm (ID=7,RASPBIAN=0)
- Command        | `apt-get -y --with-new-pkgs upgrade`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Reading package lists...

Joulinar, 2023-11-19:
Usually our updated should prevent this from happen. 
Can you check following as workaround https://github.com/MichaIng/DietPi/issues/6747#issuecomment-1808348281

pogoarte, 2023-11-19:
I don't resolve with workaround but i resolve it with:
mkdir /boot/firmware
echo '/boot    /boot/firmware   none   bind 0 0' >>/etc/fstab
systemctl daemon-reload
mount -a

Now no error when upgrade to 8.24.1

Dickesplumpel, 2023-11-19:
I had the same/similar issue. With running the workaround first the update then was succsessful.

MichaIng, 2023-11-19:
According to the logs, the `rpi-eeprom` package update was attempted before already. In this case, the automatic patch does not help. It would have been too complicated to implement it a way to handle this as well, especially since we want to actually support the new `raspi-firmware` package soon. So everyone who did run into this error after an `apt upgrade` will have to apply the workaround manually once. But at least fresh images, `dietpi-update`s and everyone who applied the live patch won't run into this in the first place, if no `apt upgrade` was done the few days before the live patch was put in place.

dixelam, 2023-11-19:
I also tried the workaround and not had succes. Maybe i try again.

MichaIng, 2023-11-19:
> I also tried the workaround and not had succes. Maybe i try again.

Logs would be helpful, if it does not work.

dixelam, 2023-11-19:
> > I also tried the workaround and not had succes. Maybe i try again.
> 
> Logs would be helpful, if it does not work.

Setting up raspi-firmware (1:1.20231024+ds-1+rpt2) ...^M
│ stat: cannot read file system information for '/boot/firmware': No such file or directory^M
│ Error: missing /boot/firmware, did you forget to mount it?^M
│ dpkg: error processing package raspi-firmware (--configure):^M
│  installed raspi-firmware package post-installation script subprocess returned error exit status 1^M
│ Errors were encountered while processing:^M
│  raspi-firmware^M
│ E: Sub-process /usr/bin/dpkg returned an error code (1)

- Date           | Sun Nov 19 13:01:51 GMT 2023
- DietPi version | v8.23.0 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bookworm (ID=7,RASPBIAN=0)
- Command        | `apt-get -y --with-new-pkgs upgrade`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
The following packages have been kept back:
  rpi-eeprom
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up raspi-firmware (1:1.20231024+ds-1+rpt2) ...
stat: cannot read file system information for '/boot/firmware': No such file or directory
Error: missing /boot/firmware, did you forget to mount it?
dpkg: error processing package raspi-firmware (--configure):
 installed raspi-firmware package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 raspi-firmware
E: Sub-process /usr/bin/dpkg returned an error code (1)

MichaIng, 2023-11-19:
I mean the logs when applying the manual workaround linked above.

dixelam, 2023-11-19:
> I mean the logs when applying the manual workaround linked above.

Maybe this will help:

The following held packages will be changed:
  rpi-eeprom
0 upgraded, 0 newly installed, 2 to remove and 0 not upgraded.
1 not fully installed or removed.
E: Held packages were changed and -y was used without --allow-change-held-packages.
root@DietPi:~# cd /tmp
root@DietPi:/tmp# curl -fo package.deb 'https://archive.raspberrypi.org/debian/pool/main/r/rpi-eeprom/rpi-eeprom_20.1-1_all.deb'
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 3245k  100 3245k    0     0  5989k      0 --:--:-- --:--:-- --:--:-- 5988k
root@DietPi:/tmp# apt install ./package.deb
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'rpi-eeprom' instead of './package.deb'
rpi-eeprom is already the newest version (20.1-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] y
Setting up raspi-firmware (1:1.20231024+ds-1+rpt2) ...
stat: cannot read file system information for '/boot/firmware': No such file or directory
Error: missing /boot/firmware, did you forget to mount it?
dpkg: error processing package raspi-firmware (--configure):
 installed raspi-firmware package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 raspi-firmware
E: Sub-process /usr/bin/dpkg returned an error code (1)

Joulinar, 2023-11-19:
you already fail on first step. Can you share following

```
apt -y purge rpi-eeprom raspi-firmware
```

dixelam, 2023-11-19:
> you already fail on first step. Can you share following
> 
> ```
> apt -y purge rpi-eeprom raspi-firmware
> ```

the command runs and the following message appears (see above):

apt -y purge rpi-eeprom raspi-firmware
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages were automatically installed and are no longer required:
  binutils binutils-aarch64-linux-gnu binutils-common libbinutils libctf-nobfd0 libctf0 libgprofng0 libjansson4
  libpci3 libpython3-stdlib pci.ids pciutils python3 python3-minimal python3.11 python3.11-minimal
Use 'apt autoremove' to remove them.
The following packages will be REMOVED:
  raspi-firmware* rpi-eeprom*
The following held packages will be changed:
  rpi-eeprom
0 upgraded, 0 newly installed, 2 to remove and 0 not upgraded.
1 not fully installed or removed.
E: Held packages were changed and -y was used without --allow-change-held-packages.

MichaIng, 2023-11-19:
Ah, because the package was set on hold already. Do this:
```sh
G_AGP rpi-eeprom raspi-firmware
```

dixelam, 2023-11-19:
> Ah, because the package was set on hold already. Do this:
> 
> ```shell
> G_AGP rpi-eeprom raspi-firmware
> ```

That WORKED!! Thanks very much!

MichaIng, 2023-11-19:
Great, I'll mark this issue as closed. I also updated the pinned again the other issue with the workaround.

closed 2023-11-19 by MichaIng