MichaIng/DietPi #6765: v8.24 upgrade error

onurcoskun14, 2023-11-20:
#### Details:
- Date           | Mon Nov 20 11:53:22 GMT 2023
- DietPi version | v8.23.3 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bookworm (ID=7,RASPBIAN=0)
- Command        | `apt-get -y --with-new-pkgs upgrade`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Reading package lists...
```

Joulinar, 2023-11-20:
Can you check this issue if it fits https://github.com/MichaIng/DietPi/issues/6763#issue-2000757265

onurcoskun14, 2023-11-20:
I tried  `sudo G_AGP rpi-eeprom raspi-firmware`
gave this output: sudo: G_AGP: command not found

Joulinar, 2023-11-20:
Can you try to login as user `root`?

onurcoskun14, 2023-11-20:
Worked thank you.

Joulinar, 2023-11-20:
Ok I'm going to close the issue. Feel free to reopen if needed

closed 2023-11-20 by Joulinar