MichaIng/DietPi #6766: Error Update 8.24.0

2pponteggi, 2023-11-20:
#### Details:
- Date           | Mon Nov 20 20:16:46 CET 2023
- DietPi version | v8.24.0 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux DietPiMao 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bookworm (ID=7,RASPBIAN=0)
- Command        | `apt-get -y --allow-change-held-packages install ./package.deb`
- Exit code      | 100
- Software title | DietPi-Patch
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Reading package lists...
Building dependency tree...
Reading state information...
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 vaultwarden : Depends: libc6 (>= 2.36-9+deb12u3) but 2.31-13+rpt2+rpi1+deb11u7 is to be installed
               Depends: openssl (>= 3.0.11-1~deb12u2) but 1.1.1w-0+deb11u1+rpt1 is to be installed
               Depends: libssl3 (>= 3.0.11-1~deb12u2) but it is not installable
E: Unable to correct problems, you have held broken packages.
```

Thank You

MichaIng, 2023-11-20:
Hmm, your distro version is shown as "Bookworm", but your APT lists are Bullseye list, it seems. Did you recently try to upgrade from Bullseye to Bookworm?

Can you show the output of the following commands:
```sh
cat /etc/os-release
cat /etc/apt/sources.list
dpkg -l base-files
```

2pponteggi, 2023-11-20:
Hi, I usually run dietpi-update for updates and dietpi-config for software.. I'm a Linux rookie...

root@DietPiMao:~# cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

root@DietPiMao:~# cat /etc/apt/sources.list
deb https://deb.debian.org/debian/ bullseye main contrib non-free
deb https://deb.debian.org/debian/ bullseye-updates main contrib non-free
deb https://deb.debian.org/debian-security/ bullseye-security main contrib non-free
deb https://deb.debian.org/debian/ bullseye-backports main contrib non-free

root@DietPiMao:~# dpkg -l base-files
Voluto=U (non noto)/I (installato)/R (rimosso)/P (rimosso totale)/H (in attesa)
| Stato=Non/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(nessuno)/R (reinstallazione richiesta) (Stato,Err: maiuscolo=grave)
||/ Nome           Versione     Architettura Descrizione
+++-==============-============-============-======================================
ii  base-files     11.1+deb11u8 arm64        Debian base system miscellaneous files

Joulinar, 2023-11-20:
@MichaIng this is the 2nd time I see a system reported as Bookworm but it is Bullseye still https://dietpi.com/forum/t/problem-with-installing-wordpress-through-dietpi-software/18451

2pponteggi, 2023-11-20:
I performed restore with dietpi-backup a few days ago with this procedure:
1. dietpi-update
2. dietpi-backup 1
3. flash image dietpi Empty
4. restore diet
everything works perfectly.

I make a backup now and then try to run:
sudo bash-c " $(curl-sSf 'https://raw.githubusercontent.com/MichaIng/DietPi/master/.meta/dietpi-bookworm-upgrade')"

i'll update you soon
Thank you very much

Joulinar, 2023-11-20:
It was not my aim to force you into the Debian destro upgrade. More a hint for @MichaIng 😉

2pponteggi, 2023-11-20:
root@DietPi:~# cat /etc/debian_version
11.8

MichaIng, 2023-11-20:
So far so good. And when you run this?
```sh
cat /boot/dietpi/.hw_model
/boot/dietpi/func/dietpi-obtain_hw_model
cat /boot/dietpi/.hw_model
```

2pponteggi, 2023-11-20:
now I'm updating...
do I open commands inside another shell?

MichaIng, 2023-11-20:
If you are upgrading to Bookworm already, all good. We seem to have some other cases to get the relevant output for debugging from.

2pponteggi, 2023-11-21:
I have updating to Bookworm and all works right!
Thank you so much

Joulinar, 2023-11-21:
ok I'm going to close the issue. Feel free to reopen if needed.

closed 2023-11-21 by Joulinar