MichaIng/DietPi #6769: Unable to update DietPi from v8.23.3 to v8.24.1

AdamF100001, 2023-11-20:
Reference code: 13d7588b-63bd-4089-b2d6-df575745ccad

Joulinar, 2023-11-20:
Would it be possible to fill the troubleshooting template or at least share some more information on your system?

MichaIng, 2023-11-20:
Based on the bug report upload, please see the pinned issue + fix: #6747

Crazy how many RPi 4 Bookworm users did run an `apt upgrade` within the 2 days or what, bug did not take care the thrown error until `dietpi-update` prompts it again 😄. I hope we handled everyone soon, else we must push another live patch which purges the two packages ...

AdamF100001, 2023-11-20:
### Creating a bug report/issue 

#### Required Information
- DietPi version | `cat /boot/dietpi/.version` 

G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=23
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='applied'
G_LIVE_PATCH_STATUS[2]='applied' 

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN` 

bookworm 0 

- Kernel version | `uname -a` 

Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux 

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi4) 

RPi 4 Model B (aarch64)

- Power supply used | (EG: 5V 1A RAVpower
- SD card used | (EG: SanDisk ultra) 

?

- Bug report ID | `echo $G_HW_UUID` 

13d7588b-63bd-4089-b2d6-df575745ccad 

#### Steps to reproduce

1. root@DietPi:~# dietpi-update

Joulinar, 2023-11-21:
pls check solution on the linked issue above https://github.com/MichaIng/DietPi/issues/6747#issuecomment-1808348281

AdamF100001, 2023-11-23:
This didn't fix the issue

Joulinar, 2023-11-23:
can you share following

```
apt update
```

or do you have any error message you can share as well?

AdamF100001, 2023-11-23:
root@DietPi:~# apt update
Hit:1 https://deb.debian.org/debian bookworm InRelease
Hit:2 https://archive.raspberrypi.org/debian bookworm InRelease
Hit:3 https://deb.debian.org/debian bookworm-updates InRelease
Hit:4 https://deb.debian.org/debian-security bookworm-security InRelease
Hit:5 https://deb.debian.org/debian bookworm-backports InRelease
Hit:6 https://repo.jellyfin.org/debian bullseye InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.

root@DietPi:~# apt upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n]
Setting up raspi-firmware (1:1.20231024+ds-1+rpt2) ...
stat: cannot read file system information for '/boot/firmware': No such file or directory
Error: missing /boot/firmware, did you forget to mount it?
dpkg: error processing package raspi-firmware (--configure):
 installed raspi-firmware package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 raspi-firmware
E: Sub-process /usr/bin/dpkg returned an error code (1)

Joulinar, 2023-11-23:
are you sure you execute following as describe on the linked issue?

```
G_AGP rpi-eeprom raspi-firmware
G_AGUP
G_AGI rpi-eeprom
```

AdamF100001, 2023-11-23:
root@DietPi:~# G_AGP rpi-eeprom raspi-firmware
[ INFO ] APT purge rpi-eeprom raspi-firmware, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
The following packages will be REMOVED:
  binutils* binutils-aarch64-linux-gnu* binutils-common* libbinutils*
  libctf-nobfd0* libctf0* libgprofng0* libjansson4* libpci3* pci.ids*
  pciutils* raspi-firmware* rpi-eeprom*
0 upgraded, 0 newly installed, 13 to remove and 0 not upgraded.
                                                               1 not fully installed or removed.
                      After this operation, 109 MB disk space will be freed.
(Reading database ... 29890 files and directories currently installed.)
Removing raspi-firmware (1:1.20231024+ds-1+rpt2) ...
Removing rpi-eeprom (20.2-1) ...
Removing binutils (2.40-2) ...
Removing binutils-aarch64-linux-gnu (2.40-2) ...
Removing libgprofng0:arm64 (2.40-2) ...
Removing libctf0:arm64 (2.40-2) ...
Removing libbinutils:arm64 (2.40-2) ...
Removing binutils-common:arm64 (2.40-2) ...
Removing libctf-nobfd0:arm64 (2.40-2) ...
Removing libjansson4:arm64 (2.14-2) ...
Removing pciutils (1:3.9.0-4) ...
Removing libpci3:arm64 (1:3.9.0-4) ...
Removing pci.ids (0.0~2023.04.11-1) ...
Processing triggers for libc-bin (2.36-9+rpt2+deb12u3) ...
(Reading database ... 29197 files and directories currently installed.)
Purging configuration files for binutils-common:arm64 (2.40-2) ...
Purging configuration files for raspi-firmware (1:1.20231024+ds-1+rpt2) ...
Purging configuration files for pciutils (1:3.9.0-4) ...
Purging configuration files for rpi-eeprom (20.2-1) ...
[  OK  ] APT purge rpi-eeprom raspi-firmware
root@DietPi:~# G_AGUP
[ INFO ] APT update, please wait...
Hit:1 https://deb.debian.org/debian bookworm InRelease
Hit:2 https://deb.debian.org/debian bookworm-updates InRelease
Hit:3 https://deb.debian.org/debian-security bookworm-security InRelease
Hit:4 https://deb.debian.org/debian bookworm-backports InRelease
Hit:5 https://archive.raspberrypi.org/debian bookworm InRelease
Hit:6 https://repo.jellyfin.org/debian bullseye InRelease
Reading package lists...
[  OK  ] APT update
root@DietPi:~# G_AGI rpi-eeprom
[ INFO ] APT install rpi-eeprom, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
The following additional packages will be installed:
  binutils binutils-aarch64-linux-gnu binutils-common libbinutils
  libctf-nobfd0 libctf0 libgprofng0 libjansson4 libpci3 pci.ids pciutils
Suggested packages:
  binutils-doc
Recommended packages:
  flashrom
The following NEW packages will be installed:
  binutils binutils-aarch64-linux-gnu binutils-common libbinutils
  libctf-nobfd0 libctf0 libgprofng0 libjansson4 libpci3 pci.ids pciutils
  rpi-eeprom
0 upgraded, 12 newly installed, 0 to remove and 0 not upgraded.
Need to get 10.7 MB of archives.
After this operation, 86.6 MB of additional disk space will be used.
Get:1 https://deb.debian.org/debian bookworm/main arm64 pci.ids all 0.0~2023.04.11-1 [243 kB]
Get:2 https://deb.debian.org/debian bookworm/main arm64 libpci3 arm64 1:3.9.0-4 [64.8 kB]
Get:3 https://deb.debian.org/debian bookworm/main arm64 pciutils arm64 1:3.9.0-4 [100.0 kB]
Get:4 https://deb.debian.org/debian bookworm/main arm64 binutils-common arm64 2.40-2 [2487 kB]
Get:5 https://archive.raspberrypi.org/debian bookworm/main arm64 rpi-eeprom all 20.2-1 [3489 kB]
Get:6 https://deb.debian.org/debian bookworm/main arm64 libbinutils arm64 2.40-2 [645 kB]
Get:7 https://deb.debian.org/debian bookworm/main arm64 libctf-nobfd0 arm64 2.40-2 [144 kB]
Get:8 https://deb.debian.org/debian bookworm/main arm64 libctf0 arm64 2.40-2 [79.2 kB]
[... 32 lines cut ...]
Preparing to unpack .../08-libjansson4_2.14-2_arm64.deb ...
Unpacking libjansson4:arm64 (2.14-2) ...
Selecting previously unselected package binutils-aarch64-linux-gnu.
Preparing to unpack .../09-binutils-aarch64-linux-gnu_2.40-2_arm64.deb ...Unpacking binutils-aarch64-linux-gnu (2.40-2) ...
Selecting previously unselected package binutils.
Preparing to unpack .../10-binutils_2.40-2_arm64.deb ...
Unpacking binutils (2.40-2) ...
Selecting previously unselected package rpi-eeprom.
Preparing to unpack .../11-rpi-eeprom_20.2-1_all.deb ...
Unpacking rpi-eeprom (20.2-1) ...
Setting up pci.ids (0.0~2023.04.11-1) ...
Setting up binutils-common:arm64 (2.40-2) ...
Setting up libctf-nobfd0:arm64 (2.40-2) ...
Setting up libjansson4:arm64 (2.14-2) ...
Setting up libpci3:arm64 (1:3.9.0-4) ...
Setting up libbinutils:arm64 (2.40-2) ...
Setting up libctf0:arm64 (2.40-2) ...
Setting up libgprofng0:arm64 (2.40-2) ...
Setting up pciutils (1:3.9.0-4) ...
Setting up binutils-aarch64-linux-gnu (2.40-2) ...
Setting up binutils (2.40-2) ...
Setting up rpi-eeprom (20.2-1) ...
Created symlink /etc/systemd/system/multi-user.target.wants/rpi-eeprom-update.service → /lib/systemd/system/rpi-eeprom-update.service.
Processing triggers for libc-bin (2.36-9+rpt2+deb12u3) ...
[  OK  ] APT install rpi-eeprom

Joulinar, 2023-11-23:
and now

```
apt update
apt upgrade
```

AdamF100001, 2023-11-23:
root@DietPi:~# apt update
Hit:1 https://deb.debian.org/debian bookworm InRelease
Hit:2 https://deb.debian.org/debian bookworm-updates InRelease
Hit:3 https://deb.debian.org/debian-security bookworm-security InRelease
Hit:4 https://deb.debian.org/debian bookworm-backports InRelease
Hit:5 https://archive.raspberrypi.org/debian bookworm InRelease
Hit:6 https://repo.jellyfin.org/debian bullseye InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
root@DietPi:~# apt upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

Joulinar, 2023-11-23:
now you are ready to perform `dietpi-update`

AdamF100001, 2023-11-23:
Ran

Dietpi-upgrade 

│ WARNING: Concurrent execution of DietPi-Update detected                │
│                                                                        │
│ Please check if one of the following applies:                          │
│  - This script already runs on another terminal/SSH session.           │
│  - Currently a cron or systemd background job executes the script.     │
│  - You started this script from within another DietPi program, causing │
│ a loop.                                                                │
│                                                                        │
│ Please assure that the concurrent execution has finished, before       │
│ retrying, otherwise cancel this instance.                              │
│                                                                        │
│ The following info might help:                                         │
│   45632 root     pts/0             \_ /bin/bash                        │
│ /boot/dietpi/dietpi-update                                             │
│   45663 root     pts/0                 \_ /bin/bash                    │
│ /boot/dietpi/dietpi-update                                             │
│   40154 navidro+ ?        /opt/navidrome/navidrome --configfile        │
│ /mnt/dietpi_userdata/navidrome/navidrome.toml

Joulinar, 2023-11-23:
reboot your system and try again

AdamF100001, 2023-11-23:
Updated successfully, Thanks

closed 2023-11-23 by AdamF100001