MichaIng/DietPi #6770: update DietPi from v8.23.3 to v8.24.1

moreboys, 2023-11-21:
#### Details:
- Date           | Tue Nov 21 18:56:27 AEDT 2023
- DietPi version | v8.23.3 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux uber-pi4gb 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bookworm (ID=7,RASPBIAN=0)
- Command        | `apt-get -y --with-new-pkgs upgrade`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ... dietpi-update
2. ...
#### Expected behaviour:

update DietPi from v8.23.3 to v8.24.1
#### Actual behaviour:

[FAILED] DietPi-Update | APT upgrade
#### Extra details:

- ...
#### Additional logs:
```
Reading package lists...
```

Joulinar, 2023-11-21:
pls check already existing issue https://github.com/MichaIng/DietPi/issues/6768#issuecomment-1819945577

Joulinar, 2024-01-01:
going to close this issue as we don't have any feedback from user.

closed 2024-01-01 by Joulinar