MichaIng/DietPi #6772: E: The repository 'http://repo.mysql.com/apt/debian bookworm InRelease' is not signed.

oespinozai, 2023-11-21:
reference code

4acdea54-5089-40e4-93d5-84a53255cab6

oespinozai, 2023-11-21:
Hi thanks for the swift reply. I installed that because I ran out of ideas reading guides on how to sort this:

Err:1 http://repo.mysql.com/apt/debian bookworm InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 4ndsvknsdkjvndkjsvsd  (redacted)

Joulinar, 2023-11-21:
You should remove this repository if not needed.

MichaIng, 2023-11-21:
> I installed that because I ran out of ideas reading guides on how to sort this:

The repository is the reason for the error, so there must have been another reason you installed it in the first place 😄.

If you hence need it for a particular reason, this should do it:
```sh
curl -sSf 'https://repo.mysql.com/RPM-GPG-KEY-mysql-2022' | gpg --dearmor -o /etc/apt/trusted.gpg.d/mysql.gpg
```
But if you do not actually need (a very recent) MySQL, I would remove it. To check which file contains it:
```sh
grep -l 'repo.mysql.com' /etc/apt/sources.list.d/*.list
```

oespinozai, 2023-11-21:
thank you all that solved it, im a complete noob

closed 2023-11-21 by oespinozai