MichaIng/DietPi #6775: Amiberry 5.6.4 won't start

RaffaeleV, 2023-11-22:
### Creating a bug report/issue

#### Required Information
- DietPi version 
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=24
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version: bookworm 0
- Kernel version: Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
- SBC model: RPi 400 (aarch64)
- Power supply used: 5V 1A RAVpower
- SD card used: Samsung EVO 64GB

#### Additional Information (if applicable)
- Software title: Amiberry 5.6.4
- Was the software title installed freshly or updated/migrated: installed freshly
- Can this issue be replicated on a fresh installation of DietPi? Yes
- Bug report ID: 3994a2de-fa20-45b3-93d4-b7552977c60f

#### Steps to reproduce
1. Deploy latest DietPI with Amiberry image, or install Amiberry using 'dietpi-sofware'
2. Try to start Amyberry

#### Expected behaviour
- Amiberry starts

#### Actual behaviour
- Error displayed on screen: 'error while loading shared libraries: libSDL2-2.0.so.0: cannot open shared object file: No such file or directory'

#### Extra details

- ...

Joulinar, 2023-11-22:
Does it help to install missing library?

```
apt install libsdl2-2.0-0
```

RaffaeleV, 2023-11-22:
Unfortunately, not: same error message.

Joulinar, 2023-11-22:
where do you see that error message?

Can you share the log

```
journalctl -u amiberry.service
```

RaffaeleV, 2023-11-22:
The error message appears when you try to launch Amiberry
Here's the journalctl -u amiberry.service
Nov 22 09:28:28 DietPi systemd[1]: Starting amiberry.service - Amiberry Amiga Emulator (DietPi)...
Nov 22 09:28:28 DietPi systemd[1]: Started amiberry.service - Amiberry Amiga Emulator (DietPi).
Nov 22 09:28:33 DietPi systemd[1]: amiberry.service: Main process exited, code=exited, status=1/FAILURE
Nov 22 09:28:33 DietPi systemd[1]: amiberry.service: Failed with result 'exit-code'.

Joulinar, 2023-11-22:
Usually, Amiberry should start automatically as a service. If you try to start it manually, you need to export library path before. Anyway, the service is failing as well. Something we need to check.

@MichaIng I could replicate the issue on my RPi4 64bit. However, a test on GitHub actions is successful https://github.com/MichaIng/DietPi/actions/runs/6956294985

MichaIng, 2023-11-22:
Right, compare with docs ("Starting" tab): https://dietpi.com/docs/software/gaming/#amiberry

To start it manually, also use the systemd service:
```sh
systemctl start amiberry
```
We use an optimised build of latest SDL2 versions, installed locally into the game's directory:
```
/mnt/dietpi_userdata/amiberry/lib
```

However, if that fails as well, that is bad. Last update was the first where it was possible to test all builds on GitHub (without an actual screen attached, which previously did not work), so I skipped testing the last point releases (5.6.2 => 5.6.3 => 5.6.4) on my RPi Zero W.

The process prints to TTY3. Could someone switch to it (CTRL+F3) and see whether there is some meaningful output? Alternatively start Amiberry this way on the current TTY to get all output there:
```sh
cd /mnt/dietpi_userdata/amiberry
LD_LIBRARY_PATH=/mnt/dietpi_userdata/amiberry/lib ./amiberry
```

Joulinar, 2023-11-22:
There you go

```
root@DietPi4:/mnt/dietpi_userdata/amiberry# LD_LIBRARY_PATH=/mnt/dietpi_userdata/amiberry/lib ./amiberry
error: XDG_RUNTIME_DIR is invalid or not set in the environment.
root@DietPi4:/mnt/dietpi_userdata/amiberry#
```

BTW: on RPi3B amrv7 it's failing on install already. But not related to this issue.

```
[ INFO ] DietPi-Software | APT install ./amiberry_rpi3.deb, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 amiberry : Depends: libudev1 (>= 252.17-1~deb12u1) but 252.12-1~deb12u1+rpi1 is to be installed
E: Unable to correct problems, you have held broken packages.
[FAILED] DietPi-Software | APT install ./amiberry_rpi3.deb
```

MichaIng, 2023-11-22:
> BTW: on RPi3B armv7 it's failing on install already. But not related to this issue.

This is a Raspbian system, right? Raspbian Bookworm is lagging behind Debian with this package version. I switched 32-bit Amiberry builds for RPi 2 - 4 to be done on Debian instead of Raspbian, to benefit from ARMv7 enhancements like NEON. But this means that the automatic dependency versions are based on Debian, not Raspbian. I wished they would faster sync. I will manually reduce the dependency version for now. Not sure whether its worth it to generally build on Raspbian again, given that most RPi 2/3/4 systems should be either ARMv7 or ARMv8 images, hence Debian. Let's see whether there is another report.

```
error: XDG_RUNTIME_DIR is invalid or not set in the environment.
```
Strange, this should be solved when installing `dbus` and unmasking/starting `systemd-logind`:
```sh
systemctl unmask systemd-logind
apt install dbus
systemctl start systemd-logind
```
However, I don't think it is the reason why Amiberry fails to start: https://github.com/BlitterStudio/amiberry/issues/1072

Joulinar, 2023-11-22:
> However, I don't think it is the reason why Amiberry fails to start

correct, even if I set XDG_RUNTIME_DIR, Amiberry is not starting. There is simply no output at all.

MichaIng, 2023-11-22:
@Joulinar 
I updated the dependencies for 32-bit RPi 2-4 builds, so they should install now.

@RaffaeleV 
How did you actually install Amiberry 5.6.4? I did test builds of this version already, but did not move then to stable place yet, hence `dietpi-software` should install v5.6.3. Probably v5.6.4 is even fixing it?
```sh
cd /tmp
curl -fO 'https://dietpi.com/downloads/binaries/bookworm/testing/amiberry_rpi4-64-dmx.deb'
dpkg -i amiberry_rpi4-64-dmx.deb
rm amiberry_rpi4-64-dmx.deb
```
It was released quite shortly after v5.6.3 with a number of bug fixes.

RaffaeleV, 2023-11-22:
@MichaIng 
I tried two methods:
1. By installing the DietPi image with Amiberry (https://dietpi.com/downloads/images/DietPi_RPi-ARMv8-Bookworm_Amiberry.img.xz)
2. By installing the DietPi "vanilla" image (https://dietpi.com/downloads/images/DietPi_RPi-ARMv8-Bookworm.img.xz) and then using the 'dietpi-software' command to install Amiberry

MichaIng, 2023-11-22:
> I tried two methods:

And you are sure both were resulting in Amiberry v5.6.**4**? How did you check this? Here is how to check the version from console:
```sh
LD_LIBRARY_PATH=/mnt/dietpi_userdata/amiberry/lib /mnt/dietpi_userdata/amiberry/amiberry -h
```

> Looks good on RPi4. Amiberry is starting as expected now.

You mean with the v5.6.4 testing version? Indeed the Raspbian repo is hanging in browser as well.

Joulinar, 2023-11-22:
> You mean with the v5.6.4 testing version?

ah yeah correct, new testing version is starting 👍

Joulinar, 2023-11-22:
Raspbian repo is back online. Install on RPI3B is working now as expected.

MichaIng, 2023-11-22:
The `libudev1` is still too low, so for the v5.6.4 builds I need to adjust the dependency version as well for now.

RaffaeleV, 2023-11-22:
@MichaIng 
My bad: I wrongly assumed that DietPi installed the latest stable version (5.3.4): but it installed Amiberry 5.6.3, as you said

MichaIng, 2023-11-22:
@RaffaeleV 
Okay, then at least all is as expected. Can you test the actual v5.6.4 now, to see whether it fixes the issue in your case as well?
```sh
cd /tmp
curl -fO 'https://dietpi.com/downloads/binaries/bookworm/testing/amiberry_rpi4-64-dmx.deb'
dpkg -i amiberry_rpi4-64-dmx.deb
rm amiberry_rpi4-64-dmx.deb
```

RaffaeleV, 2023-11-22:
@MichaIng
Just tried: now the v5.6.4 works like a charm!

MichaIng, 2023-11-22:
Okay great, I am just also testing on RPi Zero W, and will move the new packages in place then.

MichaIng, 2023-11-22:
Okay, same on RPi Zero W, v5.6.3 does not work, v5.6.4 does. Moving the new packages in place.

closed 2023-12-17 by MichaIng