MichaIng/DietPi #6777: DietPi-Software | Update Vaultwarden to v1.30.1

pdsakurai, 2023-11-23:
Hi!

May I request to update the binaries of Vaultwarden in your server?

Vaultwarden has a [new release, v1.30.1](https://github.com/dani-garcia/vaultwarden/releases/tag/1.30.1), containing the fix for auto-fill issue.

Thanks!

Regards,
Paul

MichaIng, 2023-11-23:
Done, you can upgrade with a reinstall:
```sh
sudo dietpi-software reinstall 183
```

closed 2023-11-23 by pdsakurai

JappeHallunken, 2024-07-10:
JFYI:
it get's now updated via apt, directly from dietpi repos
```
root@RPi4:~# apt upgrade
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
Paketaktualisierung (Upgrade) wird berechnet… Fertig
Die folgenden Pakete werden aktualisiert (Upgrade):
  vaultwarden
1 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
Es müssen 12,0 MB an Archiven heruntergeladen werden.
Nach dieser Operation werden 594 kB Plattenplatz zusätzlich benutzt.
Möchten Sie fortfahren? [J/n] J
Holen:1 https://dietpi.com/apt bookworm/main arm64 vaultwarden arm64 1.31.0-dietpi1 [12,0 MB]
```
❤️

MichaIng, 2024-07-10:
Yes, this has become much easier 🙂.

yuukiAme, 2024-08-12:
Because of the newest update v1.32 from [Vaultwarden](https://github.com/dani-garcia/vaultwarden/releases/tag/1.32.0).
I tried using `dietpi-software reinstall 183` and `apt update` but my Vaultwarden is not the latest version.

Screenshot here
[image]

and from release notes

```
Other changes

- Updated web-vault to v2024.6.2

```

MichaIng, 2024-08-12:
Done. vaultwarden can be updated easier in the meantime, since we ship it with our APT server:
```sh
apt update
apt install vaultwarden
```
I just updated our docs accordingly (not yet released to live docs).

MichaIng, 2025-05-26:
I know, just need to wait for new DietPi image builds to finish, before we can trigger new vaultwarden builds.