MichaIng/DietPi #6778: Wireguard DNS not working Since update to Latest Kernal ?

seedboxer, 2023-11-24:
After updating to the latest kernal, Wireguard can no longer pass DNS traffic.

Ping fails.

Looking for ideas how to troubleshoot

seedboxer, 2023-11-24:
fixed with 

sudo apt-get install raspberrypi-kernel-headers libmnl-dev libelf-dev build-essential git --yes
sudo reboot

closed 2023-11-24 by seedboxer

MichaIng, 2023-11-24:
On an RPi, it should work OOTB, since the WireGuard kernel module comes along with the kernel package. But without any further information about your system, it is difficult to know why you seem to need to compile it, via DKMS I assume?

seedboxer, 2023-11-25:
Just ran 

sudo apt-get install raspberrypi-kernel-headers libmnl-dev libelf-dev build-essential git --yes
sudo reboot

and it worked.

After kernel update it stopped working unsure why also.

But left this here to help others. thanks

MichaIng, 2023-11-26:
Then I'll add for others: Installing RPi kernel headers usually does not fix a WireGuard startup issue, especially not on RPi. This is a very specific case, with a previously manually (or via DKMS) compiled WireGuard kernel module, while RPi ships a compiled module already from Buster on. If WireGuard fails to start, check service logs to find out why:
```sh
journalctl -u wg-quick@wg0.service
```