MichaIng/DietPi #6779: No more /sys/class/leds/led0/trigger after update from DietPi 8.10.2 to DietPi v8.24.1

DrCWO, 2023-11-24:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=24
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
bullseye 0

- Kernel version | `uname -a`
Linux rooExtend 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
RPi 4 Model B (aarch64)

- Power supply used | (EG: 5V 1A RAVpower)
Officeal Raspberry Pi power supply 

- SD card used | (EG: SanDisk ultra)
Verbatim 16GB

#### Additional Information (if applicable)
I use the `/sys/class/leds/led0/trigger` to enable my own handling of the Raspberry PI LEDs.
But the path is no more available. What can be done?

#### Steps to reproduce
```
ls -al /sys/class/leds/led0/trigger
ls: cannot access '/sys/class/leds/led0/trigger': No such file or directory
```

#### Expected behaviour
```
 ls -al /sys/class/leds/led0/trigger
-rw-r--r-- 1 root root 0 Nov 24 14:41 /sys/class/leds/led0/trigger
```

#### Actual behaviour
```
ls -al /sys/class/leds/led0/trigger
ls: cannot access '/sys/class/leds/led0/trigger': No such file or directory
```
#### Extra details
None

MichaIng, 2023-11-24:
The names of the LED nodes have changed with Linux 6.1, please see the content of `/sys/class/leds`. You can also use `dietpi-led_control` to setup triggers.

DrCWO, 2023-11-24:
Thank you, fixed that in my code :-)

closed 2023-11-24 by DrCWO