MichaIng/DietPi #6780: dietpi-backup issue with chown on NFS server Truenas core

mariomare22, 2023-11-25:
Hello!
I am trying to create a backup on NFS server running on truenas core settings as follows
[image]
[image]

but i keep getting on everything that is running the following. the backup with the error below I am trying to run in a dir where I did `chmod -R 777`

```
rsync: [receiver] chown "/mnt/nfs_client/dietpi/data/usr/lib/aarch64-linux-gnu/xtables/.libarpt_mangle.so.bx1lAU" failed: Operation not permitted (1)
rsync: [receiver] chown "/mnt/nfs_client/dietpi/data/usr/lib/aarch64-linux-gnu/xtables/.libebt_802_3.so.qLId1E" failed: Operation not permitted (1)
rsync: [receiver] chown "/mnt/nfs_client/dietpi/data/usr/lib/aarch64-linux-gnu/xtables/.libebt_among.so.494XXC" failed: Operation not permitted (1)
rsync: [receiver] chown "/mnt/nfs_client/dietpi/data/usr/lib/aarch64-linux-gnu/xtables/.libebt_arp.so.GDpSOL" failed: Operation not permitted (1)
rsync: [receiver] chown "/mnt/nfs_client/dietpi/data/usr/lib/aarch64-linux-gnu/xtables/.libebt_arpreply.so.s5SW6u" failed: Operation not permitted (1)
rsync: [receiver] chown "/mnt/nfs_client/dietpi/data/usr/lib/aarch64-linux-gnu/xtables/.libebt_dnat.so.Rb9bJ1" failed: Operation not permitted (1)
rsync: [receiver] chown "/mnt/nfs_client/dietpi/data/usr/lib/aarch64-linux-gnu/xtables/.libebt_ip.so.ZVyNx4" failed: Operation not permitted (1)
```

is there anything I am doing wrong?

Thanks

mariomare22, 2023-11-27:
I think issue was related to permissions on truenas dir side
```
dietpi@zero:/home/dietpi $ ls -l /usr/lib/aarch64-linux-gnu/xtables/
total 7892
-rw-r--r-- 1 root root 67792 Jan 16  2023 libarpt_mangle.so
-rw-r--r-- 1 root root 67784 Jan 16  2023 libebt_802_3.so
-rw-r--r-- 1 root root 67784 Jan 16  2023 libebt_among.so
-rw-r--r-- 1 root root 67784 Jan 16  2023 libebt_arp.so
-rw-r--r-- 1 root root 67792 Jan 16  2023 libebt_arpreply.so
-rw-r--r-- 1 root root 67792 Jan 16  2023 libebt_dnat.so
-rw-r--r-- 1 root root 67784 Jan 16  2023 libebt_ip.so
-rw-r--r-- 1 root root 67784 Jan 16  2023 libebt_ip6.so
-rw-r--r-- 1 root root 67792 Jan 16  2023 libebt_log.so
-rw-r--r-- 1 root root 67792 Jan 16  2023 libebt_mark.so
-rw-r--r-- 1 root root 67784 Jan 16  2023 libebt_mark_m.so
-rw-r--r-- 1 root root 67792 Jan 16  2023 libebt_nflog.so
-rw-r--r-- 1 root root 67784 Jan 16  2023 libebt_pkttype.so
-rw-r--r-- 1 root root 67792 Jan 16  2023 libebt_redirect.so
-rw-r--r-- 1 root root 67792 Jan 16  2023 libebt_snat.so
-rw-r--r-- 1 root root 67784 Jan 16  2023 libebt_stp.so
-rw-r--r-- 1 root root 67784 Jan 16  2023 libebt_vlan.so
-rw-r--r-- 1 root root 67792 Jan 16  2023 libip6t_DNPT.so
-rw-r--r-- 1 root root 67792 Jan 16  2023 libip6t_HL.so
-rw-r--r-- 1 root root 67792 Jan 16  2023 libip6t_NETMAP.so
-rw-r--r-- 1 root root 67792 Jan 16  2023 libip6t_REJECT.so
-rw-r--r-- 1 root root 67792 Jan 16  2023 libip6t_SNPT.so
-rw-r--r-- 1 root root 67784 Jan 16  2023 libip6t_ah.so
-rw-r--r-- 1 root root 67784 Jan 16  2023 libip6t_dst.so
-rw-r--r-- 1 root root 67544 Jan 16  2023 libip6t_eui64.so
-rw-r--r-- 1 root root 67784 Jan 16  2023 libip6t_frag.so
-rw-r--r-- 1 root root 67784 Jan 16  2023 libip6t_hbh.so
-rw-r--r-- 1 root root 67784 Jan 16  2023 libip6t_hl.so
-rw-r--r-- 1 root root 67784 Jan 16  2023 libip6t_icmp6.so
-rw-r--r-- 1 root root 67784 Jan 16  2023 libip6t_ipv6header.so
-rw-r--r-- 1 root root 67784 Jan 16  2023 libip6t_mh.so
-rw-r--r-- 1 root root 67784 Jan 16  2023 libip6t_rt.so
-rw-r--r-- 1 root root 67976 Jan 16  2023 libip6t_srh.so
-rw-r--r-- 1 root root 67792 Jan 16  2023 libipt_CLUSTERIP.so
-rw-r--r-- 1 root root 67792 Jan 16  2023 libipt_ECN.so
-rw-r--r-- 1 root root 67792 Jan 16  2023 libipt_NETMAP.so
-rw-r--r-- 1 root root 67792 Jan 16  2023 libipt_REJECT.so
-rw-r--r-- 1 root root 67792 Jan 16  2023 libipt_TTL.so
-rw-r--r-- 1 root root 67792 Jan 16  2023 libipt_ULOG.so
-rw-r--r-- 1 root root 67784 Jan 16  2023 libipt_ah.so
-rw-r--r-- 1 root root 67784 Jan 16  2023 libipt_icmp.so
-rw-r--r-- 1 root root 67784 Jan 16  2023 libipt_realm.so
-rw-r--r-- 1 root root 67784 Jan 16  2023 libipt_ttl.so
-rw-r--r-- 1 root root 67792 Jan 16  2023 libxt_AUDIT.so
-rw-r--r-- 1 root root 67792 Jan 16  2023 libxt_CHECKSUM.so
-rw-r--r-- 1 root root 67992 Jan 16  2023 libxt_CLASSIFY.so
-rw-r--r-- 1 root root 68192 Jan 16  2023 libxt_CONNMARK.so
-rw-r--r-- 1 root root 67792 Jan 16  2023 libxt_CONNSECMARK.so
-rw-r--r-- 1 root root 68992 Jan 16  2023 libxt_CT.so
lrwxrwxrwx 1 root root    12 Jan 16  2023 libxt_DNAT.so -> libxt_NAT.so
-rw-r--r-- 1 root root 67992 Jan 16  2023 libxt_DSCP.so
-rw-r--r-- 1 root root 67992 Jan 16  2023 libxt_HMARK.so
-rw-r--r-- 1 root root 67992 Jan 16  2023 libxt_IDLETIMER.so
-rw-r--r-- 1 root root 67792 Jan 16  2023 libxt_LED.so
-rw-r--r-- 1 root root 68368 Jan 16  2023 libxt_LOG.so
-rw-r--r-- 1 root root 68520 Jan 16  2023 libxt_MARK.so
lrwxrwxrwx 1 root root    12 Jan 16  2023 libxt_MASQUERADE.so -> libxt_NAT.so
-rw-r--r-- 1 root root 69592 Jan 16  2023 libxt_NAT.so
-rw-r--r-- 1 root root 67792 Jan 16  2023 libxt_NFLOG.so
-rw-r--r-- 1 root root 68392 Jan 16  2023 libxt_NFQUEUE.so
lrwxrwxrwx 1 root root    11 Jan 16  2023 libxt_NOTRACK.so -> libxt_CT.so
-rw-r--r-- 1 root root 67792 Jan 16  2023 libxt_RATEEST.so
lrwxrwxrwx 1 root root    12 Jan 16  2023 libxt_REDIRECT.so -> libxt_NAT.so
-rw-r--r-- 1 root root 67992 Jan 16  2023 libxt_SECMARK.so
-rw-r--r-- 1 root root 68392 Jan 16  2023 libxt_SET.so
lrwxrwxrwx 1 root root    12 Jan 16  2023 libxt_SNAT.so -> libxt_NAT.so
[... 33 lines cut ...]
-rw-r--r-- 1 root root 67976 Jan 16  2023 libxt_nfacct.so
-rw-r--r-- 1 root root 67784 Jan 16  2023 libxt_osf.so
-rw-r--r-- 1 root root 68168 Jan 16  2023 libxt_owner.so
-rw-r--r-- 1 root root 67784 Jan 16  2023 libxt_physdev.so
-rw-r--r-- 1 root root 67784 Jan 16  2023 libxt_pkttype.so
-rw-r--r-- 1 root root 67976 Jan 16  2023 libxt_policy.so
-rw-r--r-- 1 root root 67784 Jan 16  2023 libxt_quota.so
-rw-r--r-- 1 root root 67784 Jan 16  2023 libxt_rateest.so
-rw-r--r-- 1 root root 68168 Jan 16  2023 libxt_recent.so
-rw-r--r-- 1 root root 67784 Jan 16  2023 libxt_rpfilter.so
-rw-r--r-- 1 root root 67784 Jan 16  2023 libxt_sctp.so
-rw-r--r-- 1 root root 68552 Jan 16  2023 libxt_set.so
-rw-r--r-- 1 root root 68360 Jan 16  2023 libxt_socket.so
-rw-r--r-- 1 root root 67712 Jan 16  2023 libxt_standard.so
lrwxrwxrwx 1 root root    18 Jan 16  2023 libxt_state.so -> libxt_conntrack.so
-rw-r--r-- 1 root root 67784 Jan 16  2023 libxt_statistic.so
-rw-r--r-- 1 root root 67976 Jan 16  2023 libxt_string.so
-rw-r--r-- 1 root root 67784 Jan 16  2023 libxt_tcp.so
-rw-r--r-- 1 root root 67784 Jan 16  2023 libxt_tcpmss.so
-rw-r--r-- 1 root root 67784 Jan 16  2023 libxt_time.so
-rw-r--r-- 1 root root 67976 Jan 16  2023 libxt_tos.so
-rw-r--r-- 1 root root 67784 Jan 16  2023 libxt_u32.so
-rw-r--r-- 1 root root 67784 Jan 16  2023 libxt_udp.so
dietpi@zero:/home/dietpi $
```

MichaIng, 2023-12-22:
Sorry for the late reply. I missed that those affected files are dot files (hidden). Can you check like that:
```sh
ls -l /usr/lib/aarch64-linux-gnu/xtables/.??*
```