MichaIng/DietPi #6783: set transmission standard to more than two downloads

hanscees, 2023-11-26:
### Creating a feature request

Hi, I am running dietpi happily, thank you for the excellent distro. 
I was trying to get transmission to download more than 2 things at the time, but that failed. 
This is because of a setting you can only get to by adjustuing a file by hand

#### Describe the solution you'd like:
I want transmission to download more then 2 files at the time by default. 

vi /etc/transmission-daemon/settings.json
Than set these values

```
    "download-queue-enabled": false,
    "download-queue-size": 21,
```

Thanks!

MichaIng, 2023-11-26:
Indeed, on all SBCs we limit active downloads to 2 for all BitTorrent downloaders, in case of `x86_64` to 3. It has been like this since before I joined the project, not sure whether this is reasonable. I am open for arguments to change this and other defaults we apply, which probably were more oriented on an RPi 1 than modern SBCs. But 21 concurrent downloads IMO is too much, as those would just limit each other, at least if you do not download via bad trackers and from very few+slow seeders only, or you have a data center grade network 🙂.

However, is it a big issue to edit the config file? I assume that 90% of users will want to fine tune settings for their personal use case anyway, so I consider the default to be a failsafe value to start with only.

It is btw either `"download-queue-enabled": false,`, which disables the limit completely, or `"download-queue-size": 21,`, which sets the limit to 21. Setting both does not make so much sense 😉.

hanscees, 2023-11-28:
Hi,
I was giving examples. 
I would disable the download queue entirely by default. Transmission has all kinds of methods in the gui where you can manage traffic as a user. 

Greetings
Hans-Cees