MichaIng/DietPi #6790: dietpi upgrade failing

itsdoublearon, 2023-12-01:
- Date           | Fri Dec  1 15:35:55 PST 2023
- DietPi version | v8.23.3 (MichaIng/master)
- Image creator  | 
- Pre-image      | 
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bookworm (ID=7,RASPBIAN=0)
- Command        | `apt-get -y --with-new-pkgs upgrade`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:
I've been trying to do the update for dietpi, but it keeps failing on the eeprom

#### Additional logs:

The following packages have been kept back:
  rpi-eeprom
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
2 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up raspi-firmware (1:1.20231024+ds-1+rpt2) ...
stat: cannot read file system information for '/boot/firmware': No such file or directory
Error: missing /boot/firmware, did you forget to mount it?
dpkg: error processing package raspi-firmware (--configure):
 installed raspi-firmware package post-installation script subprocess returned error exit status 1
dpkg: dependency problems prevent configuration of rpi-eeprom:
 rpi-eeprom depends on raspi-firmware; however:
  Package raspi-firmware is not configured yet.

dpkg: error processing package rpi-eeprom (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 raspi-firmware
 rpi-eeprom
E: Sub-process /usr/bin/dpkg returned an error code (1)

srs4511351, 2023-12-02:
I believe I have the same problem.
It seems to be related to a package problem.

I sent an automated bug report: - Reference code: 88046417-78cb-471c-85bf-7af0cece78f2

```
dietpi@DietPi:~$ sudo apt upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following packages have been kept back:
  libasound2-plugins libchromaprint1 rpi-eeprom
0 upgraded, 0 newly installed, 0 to remove and 3 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] y
Setting up raspi-firmware (1:1.20231024+ds-1+rpt2) ...
stat: cannot read file system information for '/boot/firmware': No such file or 
directory
Error: missing /boot/firmware, did you forget to mount it?
dpkg: error processing package raspi-firmware (--configure):
 installed raspi-firmware package post-installation script subprocess returned e
rror exit status 1
Errors were encountered while processing:
 raspi-firmware
E: Sub-process /usr/bin/dpkg returned an error code (1)
```
----Steve

Joulinar, 2023-12-02:
Searching existing topics would guide you to https://github.com/MichaIng/DietPi/issues/6747#issuecomment-1808348281

Joulinar, 2024-01-01:
Going to close this issue as we don't have a user response.

closed 2024-01-01 by Joulinar