MichaIng/DietPi #6797: Orange Pi 5 kernel upgrade to 6.7

bigross8, 2023-12-09:
Hello, 

I have noticed that Armbian is now providing images for Orange Pi 5/5B/5 Plus with 6.7 mainline kernel. Is it possible that there would be a option to test this kernel in the next release of DietPi? Thanks.

MichaIng, 2023-12-10:
Note that Linux 6.7 has not yet been released and also Armbian hence only provides RC images, no stable ones. However, it is good to read that mainline Linux 6.7 will mostly support RK3588, respectively Orange Pi 5, so we can switch to those once Armbian switches with stable branch for Linux 6.7 or later.

Ah, sadly Linux 6.6 is current LTS. So next stable will be with Linux 6.11 or so. Hmm, we will see whether it is worth it to shop images with Armbian "edge" branch kernel builds.

squassina, 2024-05-03:
Kernel 6.1 is out, Armbian is working some images already with this kernel. Should this request be changed to match that Kernel version or should a new one be opened?

MichaIng, 2024-05-03:
See here: https://dietpi.com/forum/t/orange-pi-5-should-we-have-switched-kernel/19929
I.e. we are about to switch.

MichaIng, 2024-05-13:
If you want to upgrade to the new kernel, we added builds to our APT repo:
```sh
apt install linux-{image,dtb}-vendor-rk35xx
```
Would be great to test them. On Orange Pi 5 Plus, sadly that kernel causes an infinite udev look with 5% constant CPU load. Since it is related to one of the HDMI ports (it has 3 of them), I hope this does not happen on Orange Pi 5 non-Plus. Dammit I just put that back into the box, after testing whether the MAC address still changes at reboot 😄.

MichaIng, 2024-08-26:
The upgrade is offered with DietPi v9.7 for all RK3588 SBCs.

closed 2024-08-26 by MichaIng