MichaIng/DietPi #6798: nfs problem when trying to

peaceman1920, 2023-12-11:
#### Details:
- Date           | Mon Dec 11 09:24:18 GMT 2023
- Bug report     | 30984361-d666-4f7e-9f8a-64400189ed43
- DietPi version | v8.24.1 (MichaIng/master)
- Image creator  | 
- Pre-image      | 
- Hardware       | Native PC (x86_64) (ID=21)
- Kernel version | `Linux Dell-DietPi-server 6.1.0-13-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.55-1 (2023-09-29) x86_64 GNU/Linux`
- Distro         | bookworm (ID=7)
- Command        | `apt-get -y -eany update`
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. start dietpi-software
2. search "nfs" via searchbox
3. mark "nfs-server"
4. continue to install
#### Expected behaviour:

- nfs-server installation
#### Actual behaviour:

- "DietPi has encountered an error                                   -
- Please create a ticket: https://github.com/MichaIng/DietPi/issues -
- Copy and paste only the BLUE lines below into the ticket "
#### Extra details:

- fresh installation of DietPi
#### Additional logs:
APT update                                                                                                           │ 
              │  - Command: apt-get -y -eany update                                                                                  │ 
              │  - Exit code: 100                                                                                                    │ 
              │  - DietPi version: v8.24.1 (MichaIng/master) | HW_MODEL: 21 | HW_ARCH: 10 | DISTRO: 7                                │ 
              │  - Error log:                                                                                                        │ 
              │ Hit:1 https://deb.debian.org/debian bookworm InRelease                                                               │ 
              │ Hit:2 https://deb.debian.org/debian bookworm-updates InRelease                                                       │ 
              │ Hit:3 https://deb.debian.org/debian-security bookworm-security InRelease                                             │ 
              │ Hit:4 https://deb.debian.org/debian bookworm-backports InRelease                                                     │ 
              │ Reading package lists...                                                                                             │ 
              │ E: lzma_read: Read error (10)                                                                                        │ 
              │ E: The package lists or status file could not be parsed or opened.

Joulinar, 2023-12-11:
Some files got corrupted. Solution should be similar to this one https://github.com/MichaIng/DietPi/issues/4304#issuecomment-826382223

peaceman1920, 2023-12-11:
you're right - it worked! thousand thanks for your super highspeed answer

closed 2023-12-11 by peaceman1920