MichaIng/DietPi #6805: please update tasmoadmin

Boebbele, 2023-12-15:
Dietpi-Software Nr. 27 Tasmoadmin

It installs an old Version v3.2.0 
Newest release is 3.3.0.

The intern self update funktion in tasmoadmin v3.2.0 Webinterface dont work.

Error: 
Warning: Undefined array key "release_url" in /var/www/tasmoadmin/pages/selfupdate.php on line 20

Warning: Undefined array key "latest_tag" in /var/www/tasmoadmin/pages/selfupdate.php on line 20

Fatal error: Uncaught TypeError: TasmoAdmin\SelfUpdate::update(): Argument #1 ($releaseUrl) must be of type string, null given, called in /var/www/tasmoadmin/pages/selfupdate.php on line 20 and defined in /var/www/tasmoadmin/src/SelfUpdate.php:28 Stack trace: #0 /var/www/tasmoadmin/pages/selfupdate.php(20): TasmoAdmin\SelfUpdate->update() #1 /var/www/tasmoadmin/index.php(66): include('...') #2 /var/www/tasmoadmin/index.php(111): render_template() #3 {main} thrown in /var/www/tasmoadmin/src/SelfUpdate.php on line 28

MichaIng, 2023-12-15:
It works well here:
```console
# curl -sSfL 'https://api.github.com/repos/TasmoAdmin/TasmoAdmin/releases/latest' | mawk -F\" '/^ *"browser_download_url": ".*\/tasmoadmin_v[^"\/]*\.tar\.gz"$/{print $4}'
https://github.com/TasmoAdmin/TasmoAdmin/releases/download/v3.3.0/tasmoadmin_v3.3.0.tar.gz
```
I guess the GitHub API was offline when you installed TasmoAdmin, and `dietpi-software` should have printed a block of `[FAILED]` informing you that obtaining the latest version failed and v3.2.0 was used as fallback, the latest known version when DietPi v8.24 was released.

Boebbele, 2023-12-15:
Strange...
I did an "dietpi-software reinstall 27"
And yet it is tasmoadmin v3.2.0 installed and no 3.3.0

[image]

MichaIng, 2023-12-15:
Ah right, a reinstall won't update TasmoAdmin due to the existence of the web updater. The web updater does still not work?

Boebbele, 2023-12-15:
Yes the webupdater dont work.

Boebbele, 2023-12-15:
[image]

Boebbele, 2023-12-15:
I uninstalled tasmoadmin with dietpi-software and installed it again.
Yet i have 3.3.0

But the self webupdater dont work. Same errorlog (Screenshot)

MichaIng, 2023-12-15:
I am currently trying to replicate the error.

MichaIng, 2023-12-15:
Hmm, the web updater worked well here:
[image]
[image]

There is a setting where the URL of the TasmoAdmin server needs to be set:
[image]

It seems the browser/client does a dedicated request to trigger the update. So if you try this from outside your LAN, the default entry (IP) might not be correct. Or does this show the public hostname, based on the request hostname, automatically?

Boebbele, 2023-12-15:
the IP addresses fit. tasmoadmin runs on a headless device local in my LAN. no remote access.

Boebbele, 2023-12-15:
[image]

Boebbele, 2023-12-15:
i have additionally installed Homer and baikal an lighthtpd is everything on port 80. is that the problem?

MichaIng, 2023-12-16:
Ah, I could reproduce it with Lighttpd! A new problem with the rewrites. Found and solved the issue:
- I was expecting that rewrites require path elements passed to the backend, like `/tasmoadmin/selfupdate => /tasmoadmin/index.php/selfupdate`.
- We already recognised previously that this rewrite causes issues with Lighttpd, since it has no way to pass FastCGI variables to tell the backend that `/tasmoadmin` is still the base path. So the backend then produces asset URLs like `/tasmoadmin/index.php/resources/css/style.css`, which of course does not exist, so pages load without styles, images and scripts. Apache2 and Nginx do this automatically, respectively have ways to tell the backend about which part is the base path to generate other URLs from.
- What worked with all other pages is to just append everything as query string: `/tasmoadmin/selfupdate => /tasmoadmin/index.php?selfupdate`. I was already wondering about this, since this did not happen in the Nginx config they use for their Docker image: https://github.com/TasmoAdmin/TasmoAdmin/blob/master/.docker/rootfs/etc/nginx/nginx.conf
- I actually thought that path elements are implied in the `$args` parameter, but rereading the docs and testing turns out this really is the query string only, while the path is completely omitted in the rewrite. I first did not understand how this can even work, but it is simple: The backend takes the path/route from the original request URL, which is always passed, instead of from the rewrite URL. So the rewrite URL must only contain `/tasmoadmin/index.php`, in case appended with the query string (rarely or not used by TasmoAdmin), and no path needs to be part of that rewrite URL.

Fixed with: https://github.com/MichaIng/DietPi/commit/b345081

Please update your config like this:
```sh
curl -sSfo /etc/lighttpd/conf-available/99-dietpi-tasmoadmin.conf 'https://raw.githubusercontent.com/MichaIng/DietPi/dev/.conf/dps_27/lighttpd.tasmoadmin.3.conf'
systemctl restart lighttpd
```

Boebbele, 2023-12-16:
It works! You are the best! Thank you very much!

closed 2023-12-17 by MichaIng