MichaIng/DietPi #6806: DietPi-Backup Error after update to 8.25

whyisthisbroken, 2023-12-16:
Sp, after updating to 8.25 i cant Backup anymore - is there a way to skip this or anything else ?

- DietPi has encountered an error                                   -
- Please create a ticket: https://github.com/MichaIng/DietPi/issues -
- Copy and paste only the BLUE lines below into the ticket          -
---------------------------------------------------------------------
#### Details:
- Date           | Sat Dec 16 14:22:54 CET 2023
- Bug report     | bc55e3db-4d0c-4448-93e1-10d9948667a2
- DietPi version | v8.25.0 (MichaIng/beta)
- Image creator  |
- Pre-image      |
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux HomeNetDNS 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bookworm (ID=7,RASPBIAN=0)
- Command        | `cp -a /etc/debian_release /mnt/m.ssd/dietpi-backup//etc/debian_release`
- Exit code      | 1
- Software title | DietPi-Backup
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:
Do the Backup
- ...
#### Actual behaviour:
well - dont backup....
- ...
#### Extra details:
Reference code: bc55e3db-4d0c-4448-93e1-10d9948667a2
- ...
#### Additional logs:
```
cp: cannot stat '/etc/debian_release': No such file or directory
```
---------------------------------------------------------------------
[FAILED] DietPi-Backup | Unable to continue, DietPi-Backup will now terminate.

MichaIng, 2023-12-16:
Thanks for your report. Fixed with: https://github.com/MichaIng/DietPi/commit/fe4311e

Please fix in your case with:
```sh
apt install patch
curl -sSf https://github.com/MichaIng/DietPi/commit/fe4311e.patch | patch -p1
```

Generally, for bugs in the beta it has some value to report them in the beta issue, so other beta testers will know about it as well: #6801

In case you were not aware that you are currently using the beta, better revert back to stable/master branch:
```sh
G_CONFIG_INJECT 'DEV_GITBRANCH=' 'DEV_GITBRANCH=master' /boot/dietpi.txt
```
Release is today night or tomorrow, so you will get the update offered soon, which migrates you to the master branch.

whyisthisbroken, 2023-12-16:
Thx - the Solution works. 💯

closed 2023-12-16 by whyisthisbroken